spacetelescope/mirage #340: Add WLP4 entry to magslist_bosz_normal_mirage.new

bhilbert4, 2019-08-07:
The +4 weak lens (WLP4) contains a filter that has a similar, but slightly different throughput curve to the F212N filter. We need to add an entry to the photometric transformation file for WLP4 so that the `get_all_catalogs` function can generate reasonable catalog magnitude values for it. @KevinVolkSTScI, do you think you could do this at some point? I have the throughput curve if you need it.

And in fact, the throughput curve should be added to the repo too, alongside the throughput curves for all the other filters.

mperrin, 2019-08-07:
There are other related changes needed for adding WLP4 too, I think: 

 - Lines for WLP4 need to be added to `config/NIRCam_zeropoints.list` for both modules A and B
 - Add `"WLP4"` to `nircam_filters` variable in `catalogs/create_catalogs.make_filter_names()` 
 - Add a line for WLP4 to `config/nircam_filter_pupil_pairings.list` 

The latter two changes are easy and I've done it in a local branch, which I will PR soon. The zeropoints and photometric transformations are harder. For now I am adding code to work around those by treating WLP4 as synonymous with F212N in those contexts, which is not correct but at least lets me proceed with test & development for now.

KevinVolkSTScI, 2019-08-09:
I have been on vacation and will be away again next week, but after I come back I can add this filter to the NIRCam simulations.  I need the photon conversion efficiency function to do the calculation.  Although if the filter profiles are only slightly different then using F212N would probably work well enough.

bhilbert4, 2020-05-08:
Returning to this issue. I have the PCE curve for WLP4. @KevinVolkSTScI I'll email it to you.

I realize that Mirage also does not support the +/-8 weak lenses. I have PCE curves for WLP8 in combination with all the allowed filters, and the PSFs are all already in the gridded PSF library. There would be a little work to do with the zeropoints file. Either a pupil column would have to be added, or some special look-up names created when WLP/M8 is used (e.g. 'WLP8+F150W'). Maybe this should be pushed to a separate issue here.

KevinVolkSTScI, 2020-05-08:
Hi Bryan,

  You may as well send me all the photon conversion efficiency curves and I will add them to my program.  I doubt I can log on to github right now because my computer has been sent off to STScI to get some help from the IT people and I am using a loaner machine.  Hence replying by e-mail rather than in the issue itself.

Kevin