spacetelescope/mirage #343: WCS PC matrix sign error

Skyhawk172, 2019-08-08:
Following up on what @laurenmarietta reported regarding creating mosaics with MIRAGE outputs, I believe (from testing) all that needs fixing are the signs for the PC matrix elements, which can be found in /utils/set_telescope_pointing_separated.py. They should be as follows:

    fheader['PC1_1']  = np.cos(pa_aper_deg * D2R)
    fheader['PC1_2'] = np.sin(pa_aper_deg * D2R)
    fheader['PC2_1']  = -np.sin(pa_aper_deg * D2R)
    fheader['PC2_2'] = np.cos(pa_aper_deg * D2R)

I am curious, however, as to how the pipeline assigns WCS information and whether the mosaic from these outputs is correctly laid out. I don't know if Lauren did that test, but we should check.

bhilbert4, 2019-08-08:
Thanks for tracking that down. The JWST calibration pipeline ignores the PC matrix, which is why mosaics created using the level 3 pipeline have all looked ok. I'll implement this fix shortly.

bhilbert4, 2020-02-12:
No fix was needed in Mirage. There was a bug in ginga/QUIP.

closed 2020-02-12 by bhilbert4