spacetelescope/mirage #367: Background level not scaled properly in WFSS dispersed images

camipacifici, 2019-09-19:
I created different simulated images running `wfss_simulator.WFSSSim` and specifying different levels of background in the yaml files, but the outputs are all the same. So I went back to the code and found that in mirage/wfss_simulator.py the background is included at `disp_seed.finalize(Back=background_file)`. This function lives in NIRCAM_Gsim/grism_seed_disperser.py and could accept a `BackLevel` argument, which is however not specified in mirage/wfss_simulator.py. Hence, the images all have the same background, while they should reflect what is specified in the yaml file.

bhilbert4, 2019-09-19:
If the background parameter in the input yaml file is a number, then we can use this and we're done (although that number is actually defined as the background signal in e-/sec/pixel across all pixels, rather than the maximum. But for now maybe that's close enough). 

But what if the background value in the yaml file(s) is one of 'low', 'medium', 'high'?
In this case the wfss_simulator needs to call jwst_backgrounds for the given RA, and Dec. I think it probably should be treated just like the imaging case. jwst_backgrounds returns an array of background signal vs wavelength for each day of the year. For imaging I loop over the days of the year, and multiply the background spectrum times the filter throughput curve. Then I integrate over the bandpass. Once this is done for all days, I take the 10th percentile version as 'low', 50th as 'medium', and 90th as 'high'.

For dispersed data, @NorPirzkal says he created the background files using LOW, and recommends scaling the file contents up by the ratio of MEDIUM/LOW or HIGH/LOW.

Mirage has a function that calls jwst_backgrounds, but it is a method of the Catalog_seed class. It might take some work to disentangle it. Best would probably be to make it a standalone function that catalog_seed_image and wfss_simulator both call.

bhilbert4, 2019-09-19:
To deal with the peak vs median background:

If the input is low, medium, high:
Call jwst_backgrounds, determine low, medium, high values, and scale by the ratio as described above.
If the input is a number (which is assumed to be the median background):
Call jwst_backgrounds, determine low, medium, high values. Calculate the median and max of the background image from the background file. BackValue is then the background rate specified in the yaml file multiplied by the ratio of the max/median values from the background image.

bhilbert4, 2019-10-03:
Nor is working on a new function to handle this problem. During our last conversation he said he'd write some code that takes a background spectrum (presumably from jwst_backgrounds, based on an input pointing and "high", "medium", "low") and produces a 2D background image.

PR #399 (2019-11-25): WFSS background scaling
This PR makes changes to the background signal in WFSS observations. 

- For NIRISS, the existing background files are scaled according to the user-input "low", "medium", "high" levels. Currently no scaling is done.

- For NIRCam, a 2D background image is computed on the fly using `jwst_backgrounds`. Users can specify an observation date, or use "low", "medium", "high". 

- The order of the yaml files in `wfss_simulator` has also been updated such that the WFSS mode file will always be first. This ensures that the direct seed image will be created from that file.

Resol [... cut]
closed 2019-11-25 by bhilbert4