spacetelescope/mirage #395: Target RA, Dec values are not propagated in WFSC observations

bhilbert4, 2019-10-28:
With #393 Mirage is now correctly populating the target RA and Dec values in the TARG_RA and TARG_DEC header keywords. This is true for imaging and wfss data, but not for WFSC data. The customized xml reader functions for the WFSC modes currently do not take in the proposal_parameter_dictionary in read_apt.py. This is where the target RA and Dec are stored. In functions for other observation modes, this dictionary is combined with the dictionary generated when reading in the observations from the xml file. 

We should update the WFSC files to make use of this dictionary so that they will also have the target RA and Dec populated.

mperrin, 2020-04-20:
@bhilbert4 is this still an open and unaddressed issue? If so I could try to take a go at it, if you'll help with with some guidance where to look...

bhilbert4, 2020-04-21:
Yes, this is still an open issue. Thanks for volunteering to take a look at it. There should only be changed needed in [read_apt_xml.py](https://github.com/spacetelescope/mirage/blob/master/mirage/apt/read_apt_xml.py). There is a separate function for reading in the information from each observing template from the APT xml file. For all templates other than those in WFSC, the `proposal_parameter_dictionary`, which contains the target RA and Dec, is integrated into the observation dictionary that is returned. So I think all that needs to be done is to add this to the functions for the WFSC templates.