spacetelescope/mirage #401: Problem when dispersing overlapping sources

camipacifici, 2019-11-06:
I wanted to create a very crowded scene with 13 sources.
[image]
The seed image looks perfectly fine, but when I try to use the disperser, it crashes with an error saying that source 15 is not in the list, while in principle the code should only count up to 13 sources.

The crash happens in `obervations.py`, where the code identifies the number of sources doing a list of the set of the ravel of the segmentation map: `all_IDs = np.array(list(set(np.ravel(self.seg))))`.
[image]
Some sources are added together when the stars and galaxies segmentation maps are added and the counts go up to 16 missing 4 and 14.

I see two problems here:
- Summing up the stars and galaxies segmentation maps is not probably the best idea when having overlapping sources;
- Also when galaxies overlap, there does not seem to be a way to keep track of which sources contribute to a specific pixel. The segmentation map seems to keep only a single number in overlapping regions.

bhilbert4, 2019-11-06:
I'll work on making the summing of the star and galaxy segmentation maps smarter. Currently observations.py in NIRCAM_Gsim does not allow for more than one source to contribute to a pixel. So for the short-term fix I will prioritize point sources over galaxies when adding those maps together. In the case where a point source and galaxy overlap, the pixel will be "given" to the point source. Similarly, when adding the segmentation map associated with the extended sources, I think it probably makes sense to prioritize the extended sources over the others.

For a longer-term solution, @NorPirzkal mentioned that with other software in the past, he has used a segmentation map cube, with one plane for each source. Perhaps this can be implemented in observations.py at some point.

bhilbert4, 2019-11-06:
A better solution than prioritizing and summing the various segmentation maps: reorganize Mirage to call the disperser once for each class of object. Then sum the dispersed seed images.

bhilbert4, 2019-11-06:
Best option for Mirage modification:

Make the seed image and segmentation map associated with each class of object a class variable so that they can be accessed from the Catalog_seed instance. Then call the disperser on each one separately within wfss_simulator.py. Finally, sum the separate dispersed seed images.

This solves the problem of overlapping sources between object classes (e.g. an overlapping point source and galaxy), however it does not solve the problem of overlapping sources within one class of object (i.e. 2 overlapping point sources).

PR #402 (2019-11-08): Overlapping wfss source types
This PR tweaks the way that the WFSS simulator code works, in order to calculate the correct dispersed signal in areas of the detector where sources **with different source types (e.g. a star and a galaxy)** overlap.  Previously the direct seed image and segmentation map were fed into the disperser. However, in cases where objects of different types overlapped, the segmentation map was being improperly calculated. In that case the object numbers from the two objects were summed. This was later interpreted as a new object number by the disperser. 

With this fix, w [... cut]
closed 2019-11-08 by bhilbert4