spacetelescope/mirage #422: WCS information at each dither position incorrect for NIRISS WFSS exposures

chriswillott, 2020-01-03:
When using the NIRISS WFSS template from APT to do a mirage simulation, the WCS values (CRPIX, CRVAL, V2_REF, RA_REF, etc) for all the dither positions are the same as in the first exposure. The spatial WCS should be updated for each dither based on the values in the .pointing file. This is important both for the direct imaging exposures that get resampled and combined into a single level 3 image and the grism exposures whose relative positions need to be known for further analysis.

bhilbert4, 2020-01-03:
This is an issue I just corrected (#419) earlier this week. If you pull down the latest version of the code, the WCS values should be correct. Let me know if they are not.

chriswillott, 2020-01-03:
Hi Bryan,

I can confirm the problem is fixed with the latest version.

Cheers, Chris

bhilbert4, 2020-01-03:
Great! Thanks for the confirmation!

closed 2020-01-03 by bhilbert4