spacetelescope/mirage #426: Mirage backgrounds for NIRISS appear to be incorrect and inconsistent between the imaging and WFSS  modes.

KevinVolkSTScI, 2020-01-08:
The background values added to the seed images for NIRISS imaging and NIRISS WFSS spectral observations are not consistent with the values from the ETC or the values generated directly from jwst_background.  The ratio of the backgrounds in imaging and WFSS spectral observations are also not consistent with what is expected or what a simple scaling of one value to the other would produce.  For the ETC the WFSS backgrounds are always about a constant factor smaller than the imaging background for the same set-up.
  I have used the ETC to get background values for imaging and WFSS modes.  It is difficult to determine exactly how many pixels are in the signal aperture for WFSS mode, although it looks like it is 1 pixel times twice the defined aperture "height", but for imaging the size can be directly calculated.  Comparison of these values for a few sky positions have shown that the Mirage values from the current version (installed 30 December) are not the same as the ETC values and are always much smaller.
  For one particular case, the proposed new NIRISS focus field at position RA = 85.22458 degrees and DEC = -69.5225 degrees the Mirage imaging background values in some of the filters are

F090W    medium    0.1360 ADU/s/pixel
F115W     medium    0.1875
F150W    medium     0,1677
F200W    medium    0.2239

and for the wfss mode the values are

Filter       Low         Medium       High
F090W   0.2834    0.3068       0.3235    electron/s/pixel
F115W    0.2788    0.3028        0.3164
F150W   0.2360    0.2556        0.2656
F200W   0.1941     0.2091        0.2159

One would expect that the count rate for the background in WFSS mode will be about 0.8 times the value in imaging mode.  With a gain value of 1.612 the imaging background values are generally smaller than the WFSS equivalents, although for the F200W case the imaging value is a little larger than the WFSS value.  But all these values are much lower than the ETC values which are 

Filter        Low       Medium      High
F090W   1.545        1.709        1.772    electrons/s/pixel
F115W    1.560        1.705        1.768
F150W   1.223         1.316        1.365
F200W   1.125         1.208        1.253

The Mirage WFSS background values are roughly a factor of 5 too small.
  The other oddity besides that the WFSS background is higher than the imaging one is that the ratio values between the filters are not the same for imaging and WFSS modes.  The ETC values for the F090W and F115W filters are fairly close, but in the Mirage case these are not similar.
  I wrote a code to use jwst_background.jbt to get the background spectra at a given position and then integrate over the NIRISS response to get the expected background signal values.  My code agrees with the ETC values to about 0.5%.  Hence I think the issue is in how Mirage is calculating the background values.

bhilbert4, 2020-01-08:
Shoot. I thought we had this all working correctly in #399. 
The ETC is a black box to me when it comes to the background calculations. But here's what I see when I try a NIRISS imaging case using F090W, medium background, and the coordinates for the focus field that you give above:

ETC:
Input Background Surface Brightness (MJy/sr): 0.29
Background signal: ~1.5 e/s/pix (= 0.93 ADU/s/pix assuming gain = 1.612)

Mirage:
background vs wavelength (from jwst_backgrounds): ~0.3 MJy/sr in each wavelength bin (0.5, 0.6, 0.7, …1.2, 1.3) This seems to agree with the ETC.
multiply by filter throughput, background becomes ~0.2 MJy/sr in each wavelength bin where wavelength bins are now dictated by filter throughput curve
integrate over bandpass (separately for each day): 0.0389 MJy/sr for day 0
low, medium, high = 0.037, 0.040, 0.042 MJy/sr
medium = 0.040 MJy/sr
Convert MJy/sr to ADU/sec:
photflam units in zeropoint file assumed to be erg/sec/cm^2/Angstrom (This is true, right?)
photflam converted to MJy at pivot wavelength (by synphot) = mjy
background value = 0.040 / (mjy / pixel_area_in steradians 
background value = 0.1359 ADU/s/pixel

Perhaps the issue has to do with integrating over the bandpass? There's about a factor of 7 difference between the integrated background signal in MJy/sr compared to the background value in each wavelength bin as supplied by jwst_backgrounds.

bhilbert4, 2020-01-08:
For WFSS, if I'm reading the code correctly, Mirage opens the background file, scales it such that the sigma-clipped mean is 1.0, and then multiplies it by a scaling factor. It looks like this scaling factor is currently the low/medium/high background value at the pivot wavelength of the crossing filter, as returned by jwst_backgrounds. I don't see any code in there to multiply the background value by the throughput of the crossing filter + grism, nor translate from MJy/sr to e-/s. It seems like both of these should be done, since the normalized background image is then multiplied by the scaling factor.

KevinVolkSTScI, 2020-01-08:
I do not understand why you are integrating the spectrum over the bandpass in MJy/ster and then treating the result as MJy/ster at the pivot wavelength.  If you really want to do this type of thing you will have to get the mean signal through the filter by dividing the integration over the spectrum by the integration over the filter profile itself.  The factor of 5 to 7 disagreement is probably because you are not allowing for the integration over the filter profile.

It would be better to convert MJy/ster to electrons/second/micron/pixel and then integrate over wavelength to get electrons/second/pixel.   That is what my code is doing.  Converting from surface brightness to electrons/second/micron/pixel is done via
    
photon_total = otearea * filterresponse * total * (1. / h) * 1.e-20 * pixelarea / wavelengths

where total is the total background spectrum from the jwst_background package interpolated to the filter response wavelength grid (wavelengths, with filterresponse being the photon conversion efficiency function for the instrument/filter).  The constants are h for Planck's constant, pixel area for the pixel solid angle in steradians, and otearea for the JWST mirror area, 25.326 square meters.  Integrating over photon_total then gives the background signal in electrons/s/pixel.

For NIRISS the WFSS background level can be just set to 0.8 times the imaging background in the blocking filter, so one does not need to fiddle with the grism orders.

bhilbert4, 2020-01-08:
Ok, I think I have it working. I implemented both methods to make sure they agreed. With the way the code is structured at the moment, it's much easier to do the integration of the filter bandpass and divide the integrated background value by that. For my test case (F090W, focus field pointing), I'm getting background values that agree well with what the ETC finds. I'll switch over the WFSS mode to use this same function to get the scaling factor and multiply that by 0.8 before multiplying it by the normalized background image.

Mirage

|                 |   low   |  medium  |  high |
|----------|-------|----------|------|
|  MJy/sr    | 0.291 |   0.315     | 0.332|
|   e-/sec   | 1.592 |    1.722    |  1.813 |

ETC

|                 |   low   |  medium  |  high |
|----------|-------|----------|------|
|  MJy/sr    | 0.29   |   0.32       | 0.33 |
|   e-/sec   |  1.59   |    1.76      | 1.81   |

KevinVolkSTScI, 2020-01-08:
Strictly speaking it is better to do it via changing to photon flux per second and integrating, but as long as the background spectrum is fairly smooth the differences should be small and averaging the sky brightness first will give a fairly close approximation to the real result.  There would be issues if the background spectrum had a lot of line structure, but with the low resolution of the background model there should not be any significant errors.  Given the uncertainties in the background itself this should be fine.

PR #430 (2020-01-22): Adjustments to bring backgrounds in line with ETC
This PR makes some adjustments to the way background levels are calculated, in order to bring those backgrounds in-line with the ETC. 

- Backgrounds for NIRISS WFSS are now explicitly 80% of those for imaging mode

- When calculating the background value from a background spectrum, Mirage integrates the spectrum over the filter bandpass, and divides this by the integral of the filter bandpass itself. This should give acceptable results unless the background spectrum is not smooth. Currently that is never the case in jwst_backgrounds.

- Fixed  [... cut]
closed 2020-01-22 by bhilbert4