spacetelescope/mirage #427: Update Besancon-related code

bhilbert4, 2020-01-08:
The team that maintains the Besancon model have significantly changed the way that users run the model and get their results. Basically, astroquery is broken (I reported this to them several months ago). With the updated model interface, a user must:

1) create an account on https://model.obs-besancon.fr/modele_home.php
2) log in and either populate a webform by hand, or download the python script created by the team and call that from the command line.
3) When the results are ready, you get an email with a link to an ftp site from which you can get the table of your results

I could conceivably create a modified version of their python program that would allow users to submit their query programmatically, but I don't see a way around having to click the ftp link in the email and download the results. 

So, I think the Besancon-related code in Mirage should be updated such that it takes a user-provided catalog conforming to the format of the table retrieved from the ftp site. From that point on, I don't think Mirage will need any changes. In this way, if Mirage users have some other method of creating or obtaining a table of stars with V, V-J, V-K, etc, they can still use the Besancon-related code in Mirage to turn that table into a Mirage-formatted catalog with source magnitudes that have been converted to the JWST filters of interest.

KevinVolkSTScI, 2020-01-08:
I agree that given the changes the users would need to do the query and then provide the file to Mirage.  The only question that comes to mind is whether we want to make it a strict format for the input catalogue or whether we want to generalize it a bit to allow other types of input that is similar?  I am thinking of isochrones from population fitting or input from some other star count model.  On the other hand, maybe that is work we should leave to the user if they are so inclined.

bhilbert4, 2020-01-08:
In the `translate_besancon` function within `createa_catalogs.py`, we currently assume the input catalog has columns for RA, Dec, V,  V-J, K-J, J-H, and J-L. The columns that go into the translation function are V, J, H,  and K. So we could split `translate_besancon` into two functions: the first takes a Besancon-formatted table and calculates V, J, H, and K, and the second function takes a table with VJHK and does the actual translation. That way users can use whatever tools they like to generate a catalog, and as long as it has VJHK, it can be used.

On a related note, there will be a mission office-imposed software freeze during the call for proposals. So any fixes that we want to have in place for people who are creating a cycle 1 proposal need to be completed and merged by Jan 23. After that we can keep developing, but users won't have access to the updates until May.

bhilbert4, 2020-01-08:
Given that, we should prioritize what work will be done in the next 2 weeks.

PR #433 (2020-01-17): Besancon catalog function updates
The manner in which the Besancon model is queried has recently changed. The query via astroquery is broken. Users must now get an account at https://model.obs-besancon.fr/modele_home.php. Queries are now done using a client provided by the group that maintains the model. The query is sent to the server, which then sends the user an email when the results are ready. The user must then download the results using an ftp link in th email.

I have adjusted the Besancon-related functions in create_catalog.py to deal with this new workflow. The functions have been ma [... cut]
closed 2020-01-17 by bhilbert4