spacetelescope/mirage #428: There is already a Mirage on Pypi

bhilbert4, 2020-01-08:
Options are:

1) Rename Mirage. If we were to follow PEP standards for package naming (short, all-lowercase, no underscores or hyphens), maybe something like `miragesim`? Or maybe ignore PEP in this case and do something easier to read, like `stsci-mirage`, `mirage-sim`? 

   This has large implications for current users. There is a [PEP recipe](https://www.python.org/dev/peps/pep-0423/#how-to-rename-a-project) in place to deal with this though.

2) Keep the name Mirage, and do not put it on Pypi. This means users would not get the convenience of installing via Pypi (e.g. pip install stsci-mirage). But users could install similarly to how the [jwst](https://github.com/spacetelescope/jwst) package is currently installed:

   To get a tagged, stable version:
pip install git+https://github.com/spacetelescope/jwst@0.14.2
To get the latest development version:
pip install git+https://github.com/spacetelescope/jwst

My first instinct is to go with option 2 since users would still be able to install via a single, not-to-convoluted command, and then current users don't have to deal with updating to the newly renamed version of the package whenever they next want to update their copy of the code. This would violate the Tier 1 standards for [JWST community software](https://github.com/spacetelescope/ins-jwst-community-software) but perhaps it's worth it given the convenience for current users.

Thoughts?
@bourque @shanosborne @KevinVolkSTScI @jdavies-st @mperrin

KevinVolkSTScI, 2020-01-08:
The issue with leaving the name as "mirage" is that it might conflict with whatever is in Pypi already.  Unless we can be reasonably sure that nobody is likely to need both we may have to change the name whether we want to or not.  I took a quick look but could not determine what the current mirage in Pypi actually does.  Assuming that we are not worried about the name conflict, option 2 is probably what we want because we need the pipeline anyway and currently that is not installed directly via Pypi.  People who are using Mirage will almost always also want to be running the pipeline on the output ramps--the exception being if they only want the seed image for something--so making the installation like the pipeline installation would provide a level of consistency.

bhilbert4, 2020-01-08:
I'm confident no one will want both this Mirage and the Mirage currently on Pypi. The other Mirage is a github wrapper of some sort where the first sentence of the readme is "Do not clone."

As for the pipeline, once it it officially released, it will be placed on Pypi and installed via `pip install jwst`. At that point I believe that only if you want a version older than the latest released version would you install using something like: pip install `git+https://github.com/spacetelescope/jwst@0.14.2`

jdavies-st, 2020-01-08:
I like `miragesim` or `mirage_sim`.  It can't be `mirage-sim`, as python package names cannot have dashes in them.

Late this year, the `jwst` calibration pipeline will be installable via PyPi.

I think installing via PyPi vastly simplifies things, so maybe bite the bullet and do the name change this one time?  Other name changes over the years:
```
pyfits --> astropy.io.fits
sphere --> spherical_geometry
finf --> pyasdf --> asdf
```
etc.

KevinVolkSTScI, 2020-01-08:
If the pipeline is going towards installation with Pypi then that makes option (1) a bit more attractive.  I am not really enthusiastic about a name change for mirage at this point, but that would simplify things for the users.  I would vote for miragesim if we have to change the name.

mperrin, 2020-01-08:
The other mirage on github is a nothing: it only has an 0.0.1 release, from five years ago, and no description of what it does or why. We should definitely NOT change this project’s name because of that. It seems ridiculous that a defunct near-zero of a project could just squat on that name indefinitely. 

As far as I know, conda is our preferred main distribution channel. Pypi is a backup older method that we support for wider compatibility but conda is the main one. 

Seems like the simplest method is to keep the name as is. I think we can work around the pypi name collision in a variety of ways, such as having this on pypi under a different name but keeping the actual package name as mirage. Meanwhile PEP-581 provides a path for requesting name transfer on pypi for defunct packages: https://www.python.org/dev/peps/pep-0541/
So

bhilbert4, 2020-01-10:
Problem solved. I emailed the author and he said that he hasn't used that package in a couple years and that he was happy to take it down from Pypi. He did that this afternoon, and I quickly uploaded mirage to claim the newly-released name. https://pypi.org/project/mirage/

closed 2020-01-10 by bhilbert4