spacetelescope/mirage #429: Some subtle issues in NIRISS imaging or WFSS simulations.

KevinVolkSTScI, 2020-01-09:
Chris Willott has pointed out that the background parameter "bkgdrate" is interpreted as ADU/second/pixel in imaging simulations but it is interpreted as electrons/second/pixel in the WFSS mode.  The comment in the imaging .yaml exampl files for this parameter is "#Constant background count rate. Number (ADU/sec/pixel) or 'high','medium','low' similar to what is used in the ETC", as in the example file imaging_test_for_notebook.yaml for example.  Apparently Chris is assuming that this is true also for WFSS parameter files.  I see that the example WFSS .yaml files such as wfss_f250m_test.yaml have a comment that the value of bkgdrate is in electrons/second/pixel.  The question is whether this should just be highlighted in the documentation or whether the code should be changed to make the units the same for both cases?  It would seem to be better to make the units of bkgdrate the same in all modes.  

Related to this issue, for NIRISS there is an issue in the gain reference file in that the values are "tiled" in 20x20 pixel blocks.  This is not a concern for the pipeline where the gain is only used in the noise statistics.  When Mirage uses the gain map for WFSS simulations it introduces small discontinuities in the spectra if these cross a tile boundary, as most of them do.  One could fix this in Mirage by using a mean gain value rather than the gain map to convert ADU/s to electrons/s.  If the code goes both ways as in imaging where the background is in ADU/s and is multiplied by the gain reference file and the frame time to get electrons/frame and then converted back to ADU for the output the tiling mostly disappears.  It would disappear entirely if the non-linearity were not applied.  For WFSS the background is in electron/s/pixel and then the gain map is only used to convert to ADU for the output ramp, and so the imprint of the gain tiles is present in the simulated frames.

Personally I would rather have Mirage use the mean gain value from NIRISS the gain reference file for all cases where the values need to be converted from ADUs to electrons, since I am not certain that the structure seen in the NIRISS gain map is actually real--one does not see analogous variations in the CV3 flat field images, and the gain should really be one value per channel not dependent on the position within the area using a given output amplifier.  It may be the gain measurements for NIRCam are better than what we have for NIRISS.  Chris tells me that he has changed the Mirage code to use the median gain rather than the gain map for NIRISS to get rid of the tile effect in the simulations.

Another small issue is that the seed images for WFSS simulations do not allow for the occulting spots before the dispersed spectral image is created.  Ideally all signal values need to be multiplied by the imaging flat field before the image is dispersed.  For point sources the signal can be multiplied by the flat field value at the star position.  For an extended source one would need to extract the flat field image for the postage stamp and multiply by this value before dispersing the spectrum.  The spots are small so this is not a big issue, but Chris would like to see the simulations allow for the spots.  

Actually the situation is more complex than described, because the "depth" of the occulting spots in the flat field images is not the same as the amount of signal loss one gets for a point source in the spot.  This presumably is due to scattered light from the edges of the spots being detected when a uniform illumination is applied, as opposed to the point source case.  However, using the flat field structure for the spots would at least give some reduction of the spectral signal for sources in the spots when WFSS observations are simulated.

bhilbert4, 2020-01-13:
Would it be better to have a map of the occulting spots that is multiplied in to the seed image before dispersing? Use this to set the signal in the pixels covered by the occulting spots be to ~zero (or some predefined percentage of the unocculted signal)? Assuming the spots are not present in the dispersed background file, those pixels will still get signal from the dispersed background, but if a source lands there in the direct image, the signal from the dispersed source will not be present.

Mirage currently multiplies galaxy and extended sources by the pixel area map in order to get their shapes correct. Do the flat field reference files include the effects of the pixel area map? If so it seems like multiplying the seed image by the flat would then result in double-applying the pixel area map.

KevinVolkSTScI, 2020-01-13:
The flat field images should include the pixel area effects.  So yes if the seed image were multiplied by the flat field image before being dispersed that would apply the pixel area effect twice.
One could multiply any extended images by the imaging filter flat field image before dispersing them.  That would take care of the occulting spots and the other POM features that get dispersed.
For point sources things could be handled the same way if the PSF image is dispersed.  I am not sure what Nor's code does for point sources, but come to think of it there must be a PSF image that is used somewhere in the process.

bhilbert4, 2020-01-13:
For PSFs Mirage does not multiply by the pixel area map because the PSFs from webbpsf are already distorted. So if someone were creating a WFSS simulation using only point sources, I'm not sure how to handle that.

KevinVolkSTScI, 2020-01-13:
The local distortion for NIRISS is rather small.  The uncertainties in the core PSF from WebbPSF must be larger than the tiny distortion effects.  (One should also convolve with the IPC kernel...).  Hence it would not be a problem to just treat the PSF image the same way.  Ideally one could use the undistorted PSF from WebbPSF, but it is just not worth the work in my opinion.  If the local distortion in NIRCam is also small then this can be done for both cases if you wanted to.

bhilbert4, 2020-02-03:
These were fixed in #431 (for the gain), and #430 (using the flats).

closed 2020-02-03 by bhilbert4

npirzkal, 2020-06-15:
Here are my $0.02 about this: the background level needs to be given in e-/s. We cannot use DN/S because these are not real units and are amplifier dependent values. The reason we gain correct images using 2D gain images  is to compensate for the different gain values of the amplifiers. Each region on the detector should have a single gain value, noiseless. This is because gain should not be confused with the  pixel to pixel flat field (pflat in HST parlance) which reflects the individual, relative, sensitivity of pixels. The gain and the flat-fields are separateble and this is tradionally why a gain and flat field are kept separately. The quickest and easier way to estimate the gain is to compare the noise in the amplifer quandrant (i.e stdev) to the signal. We can average the millions of pixels read by each ampliifer while doing that and this leads to gain estimates that are very precise, much more so that the flat-field coefficients. Finally, the flat field, reflect the pixel to pixel sensitivity and it is dangerous to include optical effect in the flat-field in cases where different modes suffers from different optical issues. This is actually exactly the case with WFSS: obscuration of the POM/FOV takes place BEFORE the grism disperses the light. A flat field containing these obscuration will lead to the wrong flat fielding being appliesd. We have had the very same issue with WFC3 since we have dust particules on the POM which acts like little out of focus occultors, and we have needed to take great care to make sure that the flat fields applies to the WFC3 WFSS (we assume gray/no wavelength dependence since wavelength dependence in WFC3 is <1%) do not contain the footprints of the blobs. This is described and shown in my ISR WFC3-2020-04.

KevinVolkSTScI, 2020-06-15:
I do not agree that the ADU/s values are not real units.  This is what we measure, so why say it is not a real unit?  Assuming that gain can be measured accurately conversion from ADU/s to electron/s is straightforward.  In practical terms, one has to account for effects like digitization noise in the simulations if one wants to match the actual observed noise so one needs to work in ADU/s.  But really the issue for NIRISS is that we have an anomally in the gain maps that we do not understand.  Ideally the gain is a single value per amplifier as Nor states.  But we have three high S/N gain maps of the NIRISS detector taken at different times with different set-ups, all of which show a structure across the amplifiers that is similar but which does not correspond to the flat field structure or anything else we have measured elsewhere.  So if these measurements are accurate and if we use a single mean value per amplifier we will make 10% errors in the gain and hence in the conversion to electron/s systematically as a function of pixel y value.  Alternatively if we state that the gain should be a single value per amplifier we have no explanation for the presistance of the structure across the amplifier primarily in the slow read direction.  This is more of an issue with the pipeline than with Mirage, but it makes me uneasy.  None the less, we probably have to work with the 2-D gain image even when we do not understand where the pixel-to-pixel structure comes from.  I dislike having this factor in the gain map that nobody understands.

As for the POM features, we have worked to correct for these in the NIRISS flat fields.  The occulting spots and a number of other POM features seen in the imaging flats are removed in the versions that are used for WFSS mode observations in the pipeline.  The POM features are observed to be significantly wavelength dependent from 0.9 to 2 microns.  The way we can find these is that changing the focus with the POM shifts any POM features on the detector and so with imaging flats at two focus positions we can determine which features need to be removed for the WFSS case; and there is a similar but smaller shift from one filter to another.