spacetelescope/mirage #436: Distortion of galxies/extended objects

bhilbert4, 2020-01-21:
Currently Mirage makes no effort to distort galaxies/extended objects. Point sources are distorted by WebbPSF. In order to properly distort the extended targets (after some discussion with @karllark ) it seems the best strategy would be to:

- Create the object at a very fine RA/Dec resolution (this may be a problem for extended sources that are being read in from random fits files with arbitrary pixel scales). 
- Map each RA/Dec location to the appropriate x,y location on the detector *at a supersampled resolution* using the distortion solution
- Bin the pixels back to the detector resolution. When doing this, to get the signal in a given pixel, average, rather than sum, the signals in the supersampled image.

This work will have to wait until after the software freeze.

mperrin, 2020-01-21:
Note, WebbPSF models distortion for *the PSF shape* for point sources (if you use the right extension). It does not model the distortion for their positions. I presume mirage is using the SIAF distortion polynomials to get the distortion into the astrometry for the point sources?

bhilbert4, 2020-01-21:
Yes, Mirage is already placing all sources, including point sources, in the correct location on the detector, using the SIAF polynomials. This issue is about using the distortion solution to modify the shape of the extended objects to be correct.