spacetelescope/mirage #445: "self" keyword missing from function in catalog_seed_image

aliciacanipe, 2020-01-29:
When trying to use an extended object input for Mirage, the software crashed due to a missing "self" keyword for the `add_segmentation_maps` function in line 2005 of `catalog_seed_image.py`:

`segmentation_map = add_segmentation_maps(segmentation_map, ext_segmap)`

It ran successfully once the keyword was added.

aliciacanipe, 2020-01-29:
Note that I don't have the exact error, I am submitting on behalf of Brad Whitmore.

bhilbert4, 2020-01-29:
Thanks for seeing the issue. I totally missed the 'self' yesterday when looking with Brad. I've corrected this in #444. I'll merge once the tests are done running, and then he can grab the master branch.

aliciacanipe, 2020-01-29:
Great, thanks! I'll pass along the message.

bhilbert4, 2020-01-29:
All set. The fixes are now in the master branch.

closed 2020-01-29 by bhilbert4