spacetelescope/mirage #486: Update xml parser to grab FGS exposures in Global Alignment template

bhilbert4, 2020-05-20:
The xml reader currently searches only for the NIRCam exposures in the global alignment template. However, the observing program also includes FGS exposures. The global alignment template reader in read_apt_xml.py needs to be updated to find and add these FGS exposures. 

Once the exposures have been added to the exposure dictionary, be sure that the information from the pointing file is correctly applied to the two instruments.

bhilbert4, 2020-05-21:
This will be a little more complex than I first thought. I added a few lines to get Mirage to read the FGS exposures in the GA template. But I ran into 2 issues.

1. The APT reader at the moment makes assumptions about the number of exposures associated with each exposure type (ADJUST2, CORRECT+ADJUST, etc). This information doesn't look to be in the xml file, so we should double check that these values are correct. It does appear that these numbers can be inferred from the number of rows in the pointing file, but the xml is read in prior to the pointing file.

2. Since the pointing file contains rows only for NIRCam, Mirage is getting confused when reading in the pointing file and trying to match up rows to the list of exposures from the xml file. So it seems that we will need to add some specialized bit of code to the pointing file reader to add rows for FGS.

mperrin, 2020-05-21:
For point 1 I was looking at that yesterday and the numbers are not correct - I fixed this in my local copy and didn't PR it yet. Here's what I recommend: 

```
        if GA_iteration == 'ADJUST1' or GA_iteration == 'CORRECT':
            # 3 NIRCam and 1 FGS images
            n_exp = 4
        elif GA_iteration == 'ADJUST2' or GA_iteration == 'CORRECT+ADJUST':
            # 5 NIRCam and 2 FGS
            n_exp = 7
        elif GA_iteration == 'BSCORRECT':
            # 2 NIRCam and 1 FGS
            n_exp = 3
```

Somewhat lower in the same function we'll want something like this to get the exposures assigned correctly: 

```

        # Repeat for the number of exposures
        for j in range(n_exp):
            # Add all parameters to dictionary
            if j==2 or j==5:
                # This is an FGS image as part of GA
                
                ## TODO update this for FGS parameters instead of NIRCam
                tup_to_add = (pi_name, prop_id, prop_title, prop_category,
                              science_category, typeflag, mod, subarr, pdithtype,
                              pdither, sdithtype, sdither, sfilt, lfilt,
                              rpatt, grps, ints, short_pupil,
                              long_pupil, grismval, coordparallel,
                              i_obs, j + 1, template_name, 'FGS', obs_label,
                              target_name)
            else:
                # This is a NIRCam image as part of GA

                tup_to_add = (pi_name, prop_id, prop_title, prop_category,
                              science_category, typeflag, mod, subarr, pdithtype,
                              pdither, sdithtype, sdither, sfilt, lfilt,
                              rpatt, grps, ints, short_pupil,
                              long_pupil, grismval, coordparallel,
                              i_obs, j + 1, template_name, 'NIRCAM', obs_label,
                              target_name)

```

bhilbert4, 2020-05-21:
Ah ok. Those numbers agree with the number of lines in the pointing file. So rather than generating new lines, we just need to switch the instrument/aperture name, and update the V2, V3 values in the lines corresponding to the FGS exposures.

Also, my quick fix last night isn't quite right then. I was assuming a 1-to-1 correspondence of NIRCam to FGS exposures. So we'll need to keep track of the right order of NIRCam and FGS exposures for each GA Iteration Type.

Frustrating that it isn't all spelled out in the xml file.

mperrin, 2020-05-21:
Right. It's not in the XML because it's all hard-coded in OPGS. Here's my reference chart that I showed yesterday: 

[GA obs types reference.pdf](https://github.com/spacetelescope/mirage/files/4662759/GA.obs.types.reference.pdf)

bhilbert4, 2020-05-21:
Ok. Since you have a fix to the APT reader that looks to be largely in place, how about you create a PR for that, and I can work on a separate PR for the pointing file information update? The downside of that is that the yaml generator will not work for GA when your PR is merged, assuming that it is merged first.

mperrin, 2020-05-21:
I can make a draft PR but it's not yet fully working. In particular I don't understand enough about the exposures dictionary content well enough to be sure about how to set that up for the FGS exposures, put into the same dict as the NIRCam ones. But I'll make a draft PR for starters.

bhilbert4, 2020-05-21:
For FGS entries, it's safe to code all the filter/pupil/grism entries as 'NA'. Assuming the FGS exposures are full frame, subarr can be FGS1_FULL or FGS2_FULL. rpatt is either 'FGS' or 'FGSRAPID'. mod is 'G'.

bhilbert4, 2020-05-21:
Here's the code I was using last night in the global alignment template reader. It's wrong because it adds an FGS exposure after each NIRCam exposure, but it does read in the FGS information correctly.

```
        # Find filter/readout parameters for all exposures within obs
        # There should be a NIRCam and FGS exposure pair for each observation
        ga_nircam_configs = template.findall('.//' + ns + 'NircamParameters')
        ga_fgs_configs = template.findall('.//' + ns + 'FgsParameters')

        if len(ga_nircam_configs) != len(ga_fgs_configs):
          raise ValueError(("ERROR: Different number of NIRCam and FGS exposures. Something "
                            "unexpected occured with the APT xml file."))

        for nrc_conf, fgs_conf in zip(ga_nircam_configs, ga_fgs_configs):
            sfilt = nrc_conf.find(ns + 'ShortFilter').text
            try:
                lfilt = nrc_conf.find(ns + 'LongFilter').text
            except AttributeError:
                lfilt = 'F480M'
            nrc_rpatt = nrc_conf.find(ns + 'ReadoutPattern').text
            nrc_grps = nrc_conf.find(ns + 'Groups').text
            nrc_ints = nrc_conf.find(ns + 'Integrations').text

            # Separate pupil and filter in case of filter that is
            # mounted in the pupil wheel
            if ' + ' in sfilt:
                split_ind = sfilt.find(' + ')
                short_pupil = sfilt[0:split_ind]
                sfilt = sfilt[split_ind + 1:]
            else:
                short_pupil = 'CLEAR'

            if ' + ' in lfilt:
                p = lfilt.find(' + ')
                long_pupil = lfilt[0:p]
                lfilt = lfilt[p + 1:]
            else:
                long_pupil = 'CLEAR'

            fgs_grps = fgs_conf.find(ns + 'Groups').text
            fgs_ints = fgs_conf.find(ns + 'Integrations').text

        # Repeat for the number of exposures + 1
        for j in range(n_exp + 1):

            # Add all NIRCam parameters to dictionary
            tup_to_add = (pi_name, prop_id, prop_title, prop_category,
                          science_category, typeflag, mod, subarr, pdithtype,
                          pdither, sdithtype, sdither, sfilt, lfilt,
                          nrc_rpatt, nrc_grps, nrc_ints, short_pupil,
                          long_pupil, grismval, coordparallel,
                          i_obs, j + 1, template_name, 'NIRCAM', obs_label,
                          target_name)

            exposures_dictionary = self.add_exposure(exposures_dictionary, tup_to_add)
            self.obs_tuple_list.append(tup_to_add)

            # Add FGS exposure to the dictionary
            tup_to_add = (pi_name, prop_id, prop_title, prop_category,
                          science_category, typeflag, 'N/A', subarr, pdithtype,
                          pdither, sdithtype, sdither, 'N/A', 'N/A',
                          'FGSRAPID', fgs_grps, fgs_ints, 'N/A',
                          'N/A', 'N/A', coordparallel,
                          i_obs, j + 1, template_name, 'FGS', obs_label,
                          target_name)

            exposures_dictionary = self.add_exposure(exposures_dictionary, tup_to_add)
            self.obs_tuple_list.append(tup_to_add)
```

mperrin, 2020-05-21:
Thanks! I'll adapt from that. 

Do you happen to have any example code for reading the SpecialRequirements section of the XML? Which guider detector to use, 1 or 2, can be read from the GuideStarID special requirement in this case.

bhilbert4, 2020-05-21:
That would also be a new section of code. Right now Mirage ignores all special requirements. I was just frustrated by this yesterday when I had a proposal with a roll angle specified via special requirements.

bhilbert4, 2020-05-21:
I think it may be necessary to add the GA_Iteration name to the exposure dictionary. I can add a function in apt_inputs.py to adjust the pointing information if an observation uses the WfscGlobalAlignment template, but I think it will also need to know the GA_Iteration name so that it knows which entries to switch to FGS. The other option would be to make assumptions about the order of the GA_Iterations within the exposure.

bhilbert4, 2020-05-27:
Note for development (so I don't forget): To get the new BaseX and BaseY values for the FGS lines in the pointing file, use:
```
import pysiaf
nrc = pysiaf.Siaf('nircam')['NRCA3_FULL']
fgs = pysiaf.Siaf('fgs')['FGS1_FULL']   # or FGS2_FULL
basex, basey = fgs.tel_to_idl(nrc.V2Ref, nrc.V3Ref)
```
This will give you the offset between the reference locations of those two apertures in the ideal coordinate system.

bhilbert4, 2020-05-28:
@mperrin Will the FGS exposures be taken using the same detector as is used for guiding? The detector used for guiding is specified in the xml, but there is nothing specifying the detector for the FGS exposures.

mperrin, 2020-05-28:
@bhilbert4 yes exactly. 

(This is opposite of how FGS external calibration imaging works normally - but in this special case the whole point is "we expect the next FGS re-acq may fail, because we're fussing with mirror state and this may confuse the FGS since the PSFs change around, so take a diagnostic image so we can see the FGS PSFs in case we have to readjust guiding settings to debug")

mperrin, 2020-05-28:
BTW, @kjbrooks pointed me at some code in MAGIC for reading the guiding special requirements, though I did not yet get a chance to adapt that for use in MIRAGE.

bhilbert4, 2020-05-28:
I should point out that there is one special requirement that Mirage does read in, although it looks like it's not used in the function for reading the GA template. Mirage looks for the Fiducial Point Override special requirement. 
https://github.com/spacetelescope/mirage/blob/master/mirage/apt/read_apt_xml.py#L520

kjbrooks, 2020-05-28:
Yes, @shanosborne wrote an APT parser for MAGIC. I have added the function [here](https://gist.github.com/kjbrooks/6034d8f5d286c346b83ba8b2f8634d47), but I think you only need some or all of the following snippet to grab the guider information: 

`        # Open the APT XML file
        with open('{}/{}.xml'.format(apt_file_path, program_id)) as f:
            tree = etree.parse(f)

        # Pull: List of Observations for the CAR > Specific Obs > Special Requirements Info (sr)
        namespace_tag = '{http://www.stsci.edu/JWST/APT}'
        observation_list = tree.find(namespace_tag + 'DataRequests').findall('.//' + namespace_tag + 'Observation')
        observation = observation_list[obs_number - 1]  # indexes from 1
        sr = [x for x in observation.iterchildren() if x.tag.split(namespace_tag)[1] == "SpecialRequirements"][0]

        # Try to pull the Guide Star information
        try:
            gs = [x for x in sr.iterchildren() if x.tag.split(namespace_tag)[1] == "GuideStarID"][0]
        except IndexError:
            self.lineEdit_normalize.setText('')
            LOGGER.error("Master GUI: This observation doesn't have a Guide Star Special Requirement")
            raise ValueError("This observation doesn't have a Guide Star Special Requirement")

        # Pull out the guide star ID and the guider number
        gs_id = [x for x in gs.iterchildren() if x.tag.split(namespace_tag)[1] == "GuideStar"][0].text
        guider = [x for x in gs.iterchildren() if x.tag.split(namespace_tag)[1] == "Guider"][0].text

        # Account for if the guider is written as "guider 1" or "guider1"
        if not isinstance(guider, int):
            guider = guider.lower().replace(' ', '').split('guider')[1]`

bhilbert4, 2020-06-05:
@mperrin I've got a working version of the yaml_generator for Global Alignment in #492 based on a manually-edited table in place of the table from reading the xml file. I added two new columns to the table, called "GA_Iteration" and "FGS_Detector".  The downside to this is that these columns will also have to be added to read_generic_imaging_template, since that is what's used to read in the NIRCam engineering imaging template. In OTE-07, a single exposure table will be produced containing information from global alignment observations mixed in with nircam engineering observations, so both need to have the new columns.

bhilbert4, 2020-09-04:
Resolved via #488

closed 2020-09-04 by bhilbert4