spacetelescope/mirage #489: Save of the scaled WFSS background used

npirzkal, 2020-05-21:
We need to have Mirage save a copy of the actual background that was used for WFSS spectroscopy. While this happens when dispersing the background using the WFSS module, there is no record of the background that was used to generate the NIRISS simulations. Just like we have copies of the SEED images, we really need to provide copies of the noiseless dispersed and scaled backgrounds too.

bhilbert4, 2020-05-21:
It looks like the place to do this is line 281 in wfss_simulator.py.
```
clip, lo, hi = sigmaclip(background_image, low=3, high=3)
background_mean = np.mean(clip)
background_image = background_image / background_mean * scaling_factor
```

bhilbert4, 2020-05-22:
Resolved in #490. I'm not sure why this issue wasn't automatically closed when the PR was merged.

closed 2020-05-22 by bhilbert4