spacetelescope/mirage #498: Problem Downloading Reference Data

eas342, 2020-06-08:
I tried downloading the reference data today and got a `RuntimeError`. Has the server URL changed?

    In [6]: import mirage                                                                                                                                                                     
    In [7]: mirage.__version__                                                                                                                                                                
    Out[7]: '1.3.3'
 
    In [8]: download_path = '/path/to/mirage_data/' 

    downloader.download_reffiles(download_path, instrument='all', dark_type='linearized', skip_darks=False, skip_cosmic_rays=False, skip_psfs=False, skip_grism=False)
    In [5]: downloader.download_reffiles(download_path, instrument='all', dark_type='linearized', skip_darks=False, skip_cosmic_rays=False, skip_psfs=False, skip_grism=False)                
    Size of NIRCam cosmic ray library file: 1.1 Gb
    ...
    Downloading: mirage_nircam_cr_library.tar.gz
    ---------------------------------------------------------------------------
    RuntimeError                              Traceback (most recent call last)
    <ipython-input-5-12f901c8bc9c> in <module>
    ...
    ~/miniconda3/envs/mirage/lib/python3.6/site-packages/mirage/reference_files/downloader.py in download_file(url, file_name, output_directory)
        220         with requests.get(url, stream=True) as response:
        221             if response.status_code != 200:
    --> 222                 raise RuntimeError("Wrong URL - {}".format(url))
        223             with open(download_filename, 'wb') as f:
        224                 for chunk in response.iter_content(chunk_size=2048):
    
    RuntimeError: Wrong URL - https://data.science.stsci.edu/redirect/JWST/jwst-simulations/mirage_reference_files/nircam/cosmic_ray_library/mirage_nircam_cr_library.tar.gz

bhilbert4, 2020-06-08:
This has been determined to be a Box error. The problem is being worked on at the moment. I'll let you know when the work is complete.

bhilbert4, 2020-06-11:
I haven't heard anything back, but I would give this another try. I know someone was working on a fix the other day.

eas342, 2020-06-11:
Thanks for checking in. I'm still getting the Wrong URL error.

bhilbert4, 2020-06-12:
When I put https://data.science.stsci.edu/redirect/JWST/jwst-simulations/mirage_reference_files/nircam/cosmic_ray_library/mirage_nircam_cr_library.tar.gz into my browser just now, it did offer to start downloading that file. Do you see the same?

eas342, 2020-06-12:
I get the following message in my browser:

> This shared file or folder link has been removed.
> 
> Think it’s a mistake? No worries: Just email the owner or get in touch with Box support. We’re here to help. Meantime, learn more about sharing files on Box – and other features, benefits and solutions – below.

bhilbert4, 2020-06-12:
Sounds like the problem is still being worked on. Sounds like it's specifically some kind of sharing problem with Box that's bad enough that the problem has been passed on to them. Did you manage to download any of the files? I was thinking that it wouldn't be too hard to get you the raw darks, since you already have CV3 darks there. I could send you an empty fits file with a DMS-formatted header, and then you could grab the pixel data from raw darks, flip it to the correct orientation, and put it in the empty fits file.

Mirage would be slower, because it will have to linearize the raw darks before using them. I'm not sure of how else to get you these large files though.

astroarshn2000, 2020-06-15:
Hi Bryan,

I am running into the same issue wherein if I try to download the reference files for NIRCam, it gives me the `RuntimeError`. Also tried the link  https://data.science.stsci.edu/redirect/JWST/jwst-simulations/mirage_reference_files/nircam/cosmic_ray_library/mirage_nircam_cr_library.tar.gz, but no luck on that one too. Any idea as to when this is getting fixed?

bhilbert4, 2020-06-15:
Hi Arsh,

This is a problem that the folks at Box are looking into. In the meantime, Everett Schlawin is in the process of downloading the reference files, and he should be in touch about a way to share them with you.

yuanshuoastro, 2020-06-22:
Hi Mirage Team,
I still can not download the MIRAGE_DATA files. Is there any other ways to get this files? 
Thank you.

bhilbert4, 2020-06-30:
I just heard back that the issues with Box have been resolved, so you should now be able to download reference files. Let me know if you have any problems.

yuanshuoastro, 2020-07-12:
The downloader is working well now.
Thank you

closed 2020-07-12 by bhilbert4