spacetelescope/mirage #502: No IDs in grism dispersion

eas342, 2020-06-18:
I am getting the following error with the NIRCam grism dispersion. It could be an error with NIRCAM_Gsim but am not 100% sure. I start with a seed image the looks OK I think, but get a dispersed image with No IDs in the segmentation map and it halts. This was encountered in the `NIRCam_TSO_examples.ipynb` notebook but I've created a smaller easier example below.

```python
from mirage.grism_tso_simulator import GrismTSO
from mirage.seed_image import catalog_seed_image
import matplotlib.pyplot as plt
import numpy as np

gr_tso_yaml_file = 'yaml_files/jw88888001001_01101_00002_nrca5.yaml'
sed_file = 'test_grism_tso_sed_file_wasp79.hdf5'

gr_f444w = GrismTSO(gr_tso_yaml_file, SED_file=sed_file, 
                    SED_normalizing_catalog_column=None,
                    final_SED_file=None, save_dispersed_seed=True,
                    source_stamps_file=None,
                    extrapolate_SED=True, override_dark=None,
                    disp_seed_filename=None,
                    orders=["+1", "+2"])

tso_direct = catalog_seed_image.Catalog_seed()
tso_direct.paramfile = gr_tso_yaml_file
tso_direct.make_seed()

#plt.imshow(tso_direct.seed_segmap)
pinfo = gr_f444w.get_param_info()
grism_seed_object = gr_f444w.run_disperser(tso_direct.seed_file,
                                           orders=gr_f444w.orders,
                                           add_background=False, 
                                           cache=True, finalize=True)

np.allclose(grism_seed_object.final,0.0)

```

Expected Bahavior: `False` because there should be nonzero pixels.
Actual Behavior: 
<details>
  <summary> AttributeError: 'observation' object has no attribute 'IDs' (Click to expand) </summary>
  
```
From CRDS, found /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_mask_0038.fits as the badpixmask reference file.
From CRDS, found /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_superbias_0022.fits as the superbias reference file.
From CRDS, found /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_linearity_0052.fits as the linearity reference file.
From CRDS, found /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_saturation_0063.fits as the saturation reference file.
From CRDS, found /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_gain_0056.fits as the gain reference file.
From CRDS, found /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_flat_0142.fits as the pixelflat reference file.
From CRDS, found /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_ipc_0028.fits as the ipc reference file.
Found an existing inverted kernel for this IPC file: /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/Kernel_to_add_IPC_effects_from_jwst_nircam_ipc_0028.fits
From CRDS, found /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_distortion_0090.asdf as the astrometric reference file.
From CRDS, found /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_area_0015.fits as the pixelAreaMap reference file.
Warning: SED Extrapolation turned on.
POM footprint applied: -129 2163 -194 2697
We have  0 Objects
dir image: /fenrirdata1/es_tso/sim_data/mirage_tso_notebook/sim_data/jw88888001001_01101_00002_nrca5_uncal_F444W_final_seed_image.fits
Loaded /fenrirdata1/es_tso/sim_data/mirage_tso_notebook/sim_data/jw88888001001_01101_00002_nrca5_uncal_F444W_final_seed_image.fits wavelength: 4.3849 micron
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-5-05f73556091e> in <module>
      4                                            orders=gr_f444w.orders,
      5                                            add_background=False,
----> 6                                            cache=True, finalize=True)

~/other_code/mirage/mirage/grism_tso_simulator.py in run_disperser(self, direct_file, orders, add_background, background_waves, background_fluxes, cache, finalize)
    785                                extrapolate_SED=self.extrapolate_SED, SED_file=self.final_SED_file,
    786                                SBE_save=self.source_stamps_file)
--> 787         disp_seed.observation(orders=orders)
    788 
    789         # Cache the dispersion if requested. This will allow you to

~/other_code/NIRCAM_Gsim/NIRCAM_Gsim/grism_seed_disperser.py in observation(self, orders, max_split, ID)
    102 
    103         for order in self.orders:
--> 104             self.this_one[order] = Gsim_observation(self.image_seeds,self.seg_data,self.config,order=order,max_split=max_split,extrapolate_SED=self.extrapolate_SED,SED_file=self.SED_file,max_cpu=self.max_cpu,ID=ID, SBE_save=self.SBE_save,boundaries=[self.xstart,self.xend,self.ystart,self.yend])
    105             #self.this_one[order].disperse_all()
    106 

~/other_code/NIRCAM_Gsim/NIRCAM_Gsim/observations/observations.py in __init__(self, direct_images, segmentation_data, config, mod, order, plot, max_split, SED_file, extrapolate_SED, max_cpu, ID, SBE_save, boundaries)
     76 
     77         self.apply_POM()
---> 78         self.create_pixel_list()
     79 
     80         self.p_l = []

~/other_code/NIRCAM_Gsim/NIRCAM_Gsim/observations/observations.py in create_pixel_list(self)
    158             if self.SED_file==None:
    159                 self.fs[l] = []
--> 160                 for i in range(len(self.IDs)):
    161                     if d[self.ys[i],self.xs[i]]>0:
    162                         self.fs[l].append(d[self.ys[i],self.xs[i]] * photflam)

AttributeError: 'observation' object has no attribute 'IDs'
```
</details>

`mirage` development
 commit `09bee457d588a52a1cc255dfd932f9685de377f9`
`NIRCAM_Gsim` development version
commit `b90299ba99aabe4682fe680b75f808fba2814f45`

I also tried modifying NIRCAM_Gsim to proceed on with `self.IDs = []` and xoffset=yoffset=0 (https://github.com/npirzkal/NIRCAM_Gsim/pull/3/files), but it then returns a value of `True` (ie. all zeros).

bhilbert4, 2020-06-19:
Did you check the segmentation map associated with the direct seed image? I'm curious if the target it present in there. It's saved in a second extension in the file that contains the direct seed image.

eas342, 2020-06-19:
Yes, the point source appears in `plt.imshow(tso_direct.seed_segmap)`:
[image]

bhilbert4, 2020-06-19:
Hmm. But somehow the disperser thinks there aren't any targets, given the message: We have  0 Objects.
tso_direct.seed_file contains the correct seed image and segmentation map?

eas342, 2020-06-19:
Yes, the `tso_direct.seed_file` looks good withe one point source in the segmentation map and one in the image, just like the above screenshot. So this may be a NIRCAM_Gsim issue.

bhilbert4, 2020-06-19:
@NorPirzkal any ideas?

npirzkal, 2020-06-19:
which grism is this using? R or C? with the F444W filter, right? I wonder if the source is in a location which is out of the effective field of view of that combination. Also, did you check your SED file?

bhilbert4, 2020-06-19:
This is TSO data, so grism R. Good point about the fov though. The direct seed image adds 424 rows and columns around the outside of the detector. The source looks to be somewhere around row 550-600, so 125-175 rows from the bottom of the detector. @eas342, which subarray are you using? Anything more narrow than the 256x2048 and the source may be off the subarray.

eas342, 2020-06-19:
I am using all the default values from the NIRCam_TSO_examples Jupyter notebook. So Grism256 (2048x256) subarray and GRISMR. In the final image, the source should be 35 pixels from the bottom, but if it's at y=150, that should still be in the subarray. Thanks for clarifying why the map is 2896x2896.

bhilbert4, 2020-06-19:
In my original TSO dataset, I see the target in the direct seed image at about (x, y) = (1376, 459)

eas342, 2020-06-19:
@NorPirzkal , can you check the SED file (attached)? I think it would take me much longer to figure out how to unpack and investigate.
[test_grism_tso_sed_file_wasp79.hdf5.zip](https://github.com/spacetelescope/mirage/files/4805858/test_grism_tso_sed_file_wasp79.hdf5.zip)

NorPirzkal, 2020-06-19:
Looks ok. I see the ID is 99999 and wavelength range and units seem ok.

eas342, 2020-06-19:
Thanks for checking Nor. In the TSO dataset I ran from the example notebook, the target is at (x,y) = (1017,575) so something is different from the original. I also realized I was using the pipeline jwst version 0.16 so I downgraded to 0.14 and that doesn't change anything. I'm using `pysiaf 0.9.0` and `PRDOPSSOC-029`. Should I use a different version of `pysiaf`?

NorPirzkal, 2020-06-19:
I will defer to Bryan about that.

bhilbert4, 2020-06-19:
Definitely stick with the later version of the pipeline. 0.16.2 has been released, and that should have some fixes over 0.14. Pysiaf and PRDOPSSOC-29 should be ok. (I used 26 or 27 when creating my data, but I don't think there have been any updates to the location of the TSO subarrrays since then.)

Since you're running based on the notebook, you have created yaml files from scratch, right? I was worried about the pointing being wrong if you had manually edited an existing yaml file. 

Can you me send your yaml file? I can compare it to the file I used for my dataset.

eas342, 2020-06-19:
Sure, here is the YAML file and catalog sources.
[yaml_file_and_catalogs.zip](https://github.com/spacetelescope/mirage/files/4806164/yaml_file_and_catalogs.zip)

bhilbert4, 2020-06-19:
So far everything looks the same. Your tso_grism_source.cat wasn't in your zip file, but I'm guessing it looks just like mine:
# position_RA_Dec
# vegamag
# 
# 
index x_or_RA y_or_Dec Semimajor_axis_in_stellar_radii Orbital_inclination_deg Eccentricity Longitude_of_periastron Limb_darkening_model Limb_darkening_coeffs Time_units Start_time End_time Time_of_inferior_conjunction Orbital_period Transmission_spectrum nircam_f444w_magnitude nircam_f322w2_magnitude
99999 66.37090333 -30.60044722 9.37 83.3 0.0 90.0 nonlinear "0.5, 0.1, 0.1, -0.1" second 0.0 580.0 280.0 3162.24 ./transmission_spectrum.txt 9.0 9.05

eas342, 2020-06-19:
Yes, my tso_grism_source.cat is identical to this one

bhilbert4, 2020-06-19:
I sent off a note asking if there are any changes to the PRD between version 26 and 29. That's about the only difference between our cases that I see.

bhilbert4, 2020-06-19:
It looks like there was a small shift in the reference location of the NRCA5_GRISM256_F444W aperture between PRD version 25 and 26 (the current version is 29). Maybe my data are old enough that I used version 25? Even if that's true, the shift is 65 pixels in x, but no shift in y, so that doesn't seem to be the issue.

bhilbert4, 2020-06-19:
It just struck me: the disperser doesn't know anything about the aperture being used. It only works on full frame data, which is why the direct seed image is 2048x2048 plus the extra rows/cols to cover the POM. So regardless of whether the source is within or outside the subarray, the disperser should find the target and disperse it. Right @NorPirzkal ?

NorPirzkal, 2020-06-19:
Correct. Full frames only.

chriswillott, 2020-06-22:
@NorPirzkal I think this is the same POM mask bug I told you about last week. When applying the POM mask in NIRCAM_Gsim the typo in the indices removes all the sources.

bhilbert4, 2020-06-25:
@NorPirzkal  this has been fixed with your latest updates to NIRCAM_Gsim, right? This issue can be closed?

closed 2020-06-26 by bhilbert4