spacetelescope/mirage #505: NIRCam_Gsim Disperser Error: X and Y values must have atleast 2 entries

astroarshn2000, 2020-06-24:
This error is with regards to the `NIRCam_TSO_examples.ipynb` notebook provided in the examples. I was able to simulate the WASP-79b Grism observation in the F444W filter, however, am getting the following error while simulating the same Grism observation in the F322W2 filter. 
Used the following lines to simulate the observations -
```ruby
gr_f322w2 = GrismTSO(gr_tso_f322w2_yaml_file, SED_file=sed_file, SED_normalizing_catalog_column=None,
                     final_SED_file=None, save_dispersed_seed=True, source_stamps_file=None,
                     extrapolate_SED=True, override_dark=None, disp_seed_filename=None,
                     orders=["+1", "+2"])
gr_f322w2.create()
```
The following Value Error was encountered - 
```ruby
---------------------------------------------------------------------------
RemoteTraceback                           Traceback (most recent call last)
RemoteTraceback: 
"""
Traceback (most recent call last):
  File "/Users/astroarshn2000/anaconda3/envs/astro/lib/python3.6/multiprocessing/pool.py", line 119, in worker
    result = (True, func(*args, **kwds))
  File "/Users/astroarshn2000/anaconda3/envs/astro/lib/python3.6/site-packages/NIRCAM_Gsim/observations/observations.py", line 18, in helper
    p = dispersed_pixel(x0s,y0s,f,order,C,ID,extrapolate_SED=extrapolate_SED,xoffset=xoffset,yoffset=yoffset)
  File "/Users/astroarshn2000/anaconda3/envs/astro/lib/python3.6/site-packages/NIRCAM_Gsim/disperse/disperse.py", line 52, in dispersed_pixel
    f = interp1d(f0[0],f0[1],fill_value="extrapolate",bounds_error=False)
  File "/Users/astroarshn2000/anaconda3/envs/astro/lib/python3.6/site-packages/scipy/interpolate/interpolate.py", line 541, in __init__
    "least %d entries" % minval)
ValueError: x and y arrays must have at least 2 entries
"""

The above exception was the direct cause of the following exception:

ValueError                                Traceback (most recent call last)
<ipython-input-83-aec3428a7092> in <module>
      3                      extrapolate_SED=True, override_dark=None, disp_seed_filename=None,
      4                      orders=["+1", "+2"])
----> 5 gr_f322w2.create()

~/anaconda3/envs/astro/lib/python3.6/site-packages/mirage/grism_tso_simulator.py in create(self)
    238                                           background_waves=bkgd_waves,
    239                                           background_fluxes=bkgd_fluxes,
--> 240                                           finalize=True)
    241                 if not background_done:
    242                     # Background is added at the first opportunity. At this

~/anaconda3/envs/astro/lib/python3.6/site-packages/mirage/grism_tso_simulator.py in run_disperser(self, direct_file, orders, add_background, background_waves, background_fluxes, cache, finalize)
    793                 disp_seed.this_one[order].disperse_all(cache=True)
    794         else:
--> 795             disp_seed.disperse(orders=orders)
    796 
    797         # Only finalize and/or add the background if requested.

~/anaconda3/envs/astro/lib/python3.6/site-packages/NIRCAM_Gsim/grism_seed_disperser.py in disperse(self, orders, cache, trans)
    127                 self.this_one[order].disperse_all_from_cache(trans=trans)
    128             else:
--> 129                 self.this_one[order].disperse_all(cache=cache)
    130 
    131     def disperse_background_1D(self,background):

~/anaconda3/envs/astro/lib/python3.6/site-packages/NIRCAM_Gsim/observations/observations.py in disperse_all(self, cache)
    260                 self.cached_object[i]['maxy'] = []
    261 
--> 262             this_object = self.disperse_chunk(i)
    263 
    264             # if cached:

~/anaconda3/envs/astro/lib/python3.6/site-packages/NIRCAM_Gsim/observations/observations.py in disperse_chunk(self, c)
    456 
    457 
--> 458         for i,pp in enumerate(all_res, 1):
    459 
    460             if np.shape(pp.transpose())==(1,6):

~/anaconda3/envs/astro/lib/python3.6/multiprocessing/pool.py in next(self, timeout)
    733         if success:
    734             return value
--> 735         raise value
    736 
    737     __next__ = next                    # XXX

ValueError: x and y arrays must have at least 2 entries
```
Don't seem to understand why this is happening.

bhilbert4, 2020-06-24:
@NorPirzkal

bhilbert4, 2020-06-25:
Nor pushed some fixes to the disperser code yesterday that should solve this issue. Try reinstalling NIRCAM_Gsim and re-running the notebook.

astroarshn2000, 2020-06-25:
That has solved the issue. Thanks Bryan!

closed 2020-06-25 by astroarshn2000