spacetelescope/mirage #508: OSError: File 'WFSS_background.fits' already exists

eas342, 2020-06-26:
There's a small error Arsh discovered with the `NIRCam_TSO_example.ipynb` notebook.

After the F444W simulation, the WFSS_background.fits file is written and can't be overwritten. An easy solution is to put

`os.replace('WFSS_background.fits','WFSS_background_F444W.fits')` or `os.rename('WFSS_background.fits','WFSS_background_F444W.fits')` between the F444W and F322W2 simulations in the notebook.

<details>
  <summary>Click to see the full traceback</summary>
  
  ```python
Computing dispersed background for order  +1
disperse_background_1D: x-direction
1448 1448
Computing dispersed background for order  +2
disperse_background_1D: x-direction
1448 1448
Traceback (most recent call last):
  File "NIRCam_TSO_examples_wasp79.py", line 677, in <module>
    gr_f444w.create()
  File "/home/schlawin/other_code/mirage/mirage/grism_tso_simulator.py", line 240, in create
    finalize=True)
  File "/home/schlawin/other_code/mirage/mirage/grism_tso_simulator.py", line 800, in run_disperser
    background_image = disp_seed.disperse_background_1D([background_waves, background_fluxes])
  File "/home/schlawin/miniconda3/envs/mirage/lib/python3.6/site-packages/NIRCAM_Gsim/grism_seed_disperser.py", line 153, in disperse_background_1D
    fits.writeto("WFSS_background.fits",bck)
  File "/home/schlawin/miniconda3/envs/mirage/lib/python3.6/site-packages/astropy/utils/decorators.py", line 535, in wrapper
    return function(*args, **kwargs)
  File "/home/schlawin/miniconda3/envs/mirage/lib/python3.6/site-packages/astropy/io/fits/convenience.py", line 420, in writeto
    checksum=checksum)
  File "/home/schlawin/miniconda3/envs/mirage/lib/python3.6/site-packages/astropy/utils/decorators.py", line 535, in wrapper
    return function(*args, **kwargs)
  File "/home/schlawin/miniconda3/envs/mirage/lib/python3.6/site-packages/astropy/io/fits/hdu/base.py", line 372, in writeto
    checksum=checksum)
  File "/home/schlawin/miniconda3/envs/mirage/lib/python3.6/site-packages/astropy/utils/decorators.py", line 535, in wrapper
    return function(*args, **kwargs)
  File "/home/schlawin/miniconda3/envs/mirage/lib/python3.6/site-packages/astropy/io/fits/hdu/hdulist.py", line 934, in writeto
    fileobj = _File(fileobj, mode=mode, overwrite=overwrite)
  File "/home/schlawin/miniconda3/envs/mirage/lib/python3.6/site-packages/astropy/utils/decorators.py", line 535, in wrapper
    return function(*args, **kwargs)
  File "/home/schlawin/miniconda3/envs/mirage/lib/python3.6/site-packages/astropy/io/fits/file.py", line 193, in __init__
    self._open_filename(fileobj, mode, overwrite)
  File "/home/schlawin/miniconda3/envs/mirage/lib/python3.6/site-packages/astropy/io/fits/file.py", line 563, in _open_filename
    self._overwrite_existing(overwrite, None, True)
  File "/home/schlawin/miniconda3/envs/mirage/lib/python3.6/site-packages/astropy/io/fits/file.py", line 453, in _overwrite_existing
    raise OSError(f"File {self.name!r} already exists.")
OSError: File 'WFSS_background.fits' already exists.
  ```
</details>

bhilbert4, 2020-06-27:
I've passed this along as an issue in NIRCAM_Gsim, since the original error is there. I think all that needs to be done is add an overwrite=True in grism_seed_disperser.py
https://github.com/npirzkal/NIRCAM_Gsim/issues/5

bhilbert4, 2020-06-30:
Fixed in NIRCAM_Gsim.

closed 2020-06-30 by bhilbert4