spacetelescope/mirage #518: Look into skipping dark_prep for segmented observations

bhilbert4, 2020-07-15:
dark_prep already has the ability to accept the name of a previously created dark object file, in which case dark_prep is skipped. However this was implemented long ago, before the idea of segmented files. Update dark_prep to accept a list of dark object files (e.g. from a prior run of Mirage), so that it can be skipped. In the case of something like TSO observations, this could potentially save a lot of time.

bhilbert4, 2020-07-15:
@astroarshn2000 this will be of interest to you.

astroarshn2000, 2020-07-15:
This will be so helpful. Thanks, Bryan!

closed 2020-07-20 by bhilbert4

astroarshn2000, 2020-07-23:
Hi Bryan,
After this update, this is a new issue popping up for me. Seems like MIRAGE doesn't want to create new dark_prep files if I don't give it a list.
```
Running dark prep

Running dark_prep....

From CRDS, found /home/anadkarni/crds_cache/references/jwst/nircam/jwst_nircam_mask_0038.fits as the badpixmask reference file.
From CRDS, found /home/anadkarni/crds_cache/references/jwst/nircam/jwst_nircam_superbias_0022.fits as the superbias reference file.
From CRDS, found /home/anadkarni/crds_cache/references/jwst/nircam/jwst_nircam_linearity_0052.fits as the linearity reference file.
From CRDS, found /home/anadkarni/crds_cache/references/jwst/nircam/jwst_nircam_saturation_0063.fits as the saturation reference file.
From CRDS, found /home/anadkarni/crds_cache/references/jwst/nircam/jwst_nircam_gain_0056.fits as the gain reference file.
From CRDS, found /home/anadkarni/crds_cache/references/jwst/nircam/jwst_nircam_flat_0142.fits as the pixelflat reference file.
From CRDS, found /home/anadkarni/crds_cache/references/jwst/nircam/jwst_nircam_ipc_0028.fits as the ipc reference file.
Found an existing inverted kernel for this IPC file: /home/anadkarni/crds_cache/references/jwst/nircam/Kernel_to_add_IPC_effects_from_jwst_nircam_ipc_0028.fits
From CRDS, found /home/anadkarni/crds_cache/references/jwst/nircam/jwst_nircam_distortion_0090.asdf as the astrometric reference file.
From CRDS, found /home/anadkarni/crds_cache/references/jwst/nircam/jwst_nircam_area_0015.fits as the pixelAreaMap reference file.
From CRDS, found /home/anadkarni/crds_cache/references/jwst/nircam/jwst_nircam_photom_0071.fits as the photom reference file.
Requested readout pattern RAPID is valid. Using the nframe = 1 and nskip = 0
Traceback (most recent call last):
  File "NIRCam_TSO_HD209458_sine.py", line 176, in <module>
    gr_f322w2.create()
  File "/home/anadkarni/anaconda3/envs/astro/lib/python3.7/site-packages/mirage/grism_tso_simulator.py", line 497, in create
    d.prepare()
  File "/home/anadkarni/anaconda3/envs/astro/lib/python3.7/site-packages/mirage/dark/dark_prep.py", line 658, in prepare
    self.check_params()
  File "/home/anadkarni/anaconda3/envs/astro/lib/python3.7/site-packages/mirage/dark/dark_prep.py", line 100, in check_params
    raise ValueError(("WARNING: No raw nor linearized dark file given. Unable to proceed."))
ValueError: WARNING: No raw nor linearized dark file given. Unable to proceed.
```

bhilbert4, 2020-07-23:
Hmmm. You should only see this if your yaml file has None for both the raw and linearized dark entries. Is that the case?

astroarshn2000, 2020-07-23:
Yes. Probably MiRAGE isn't reading the darks right. Similar to that old issue we had.

bhilbert4, 2020-07-24:
That shouldn't be an issue any more after the fixes from earlier. Which darks do you have and which do you not have in your reference file collection?

astroarshn2000, 2020-07-25:
I have both the linearized and raw, but for some reason, MIRAGE isn't finding them automatically now.

PR #522 (2020-07-20): Allow dark_prep to be skipped if dark files exist
This PR modifies the simulator wrappers (imaging, wfss, grism_tso) such that, if the outputs from a call to dark_prep.py already exist, the user can provide these outputs to Mirage, and Mirage will then skip the call to dark_prep.py. This should help save time when testing the code, and also when repeating the creation of simulations.

These changes work both for segmented and non-segmented exposures.

Resolves #518