spacetelescope/mirage #521: Mirage is multiplying seed images by the flat field twice

bhilbert4, 2020-07-16:
@micaelabagley has found that Mirage multiplies simulations by the flat field reference file twice.

Within `catalog_seed_image.py`, the various pieces of the seed image are multiplied by the flat field before being combined into the final seed image. But then in `obs_generator.py` the flat field is multiplied again, just after using the seed image to create the exposure with all its integrations and groups.

So which one should we remove? I think we don't want to multiply the seed image by the flat. The idea of the seed image in general is that it doesn't contain detector effects (except for distortion, which needs to be included). For WFSS simulations, the seed image is provided to the disperser. The flat should be applied after dispersing since it is the dispersed light that hits the detector and is subject to the pixel-to-pixel sensitivity variations.

Is there any reason we want/need the flat applied in the seed image?

https://github.com/spacetelescope/mirage/blob/master/mirage/ramp_generator/obs_generator.py#L1239

bhilbert4, 2020-07-16:
This issue is quickly becoming complicated. NIRISS wants seed images multiplied by the flats before dispersing, because their flats contain the transmission maps of their occulters, and the occulters come before the grisms in the optical path. The final solution will be to produce transmission maps that contain the occulters and which are separate from the flats. Then the transmission maps can be applied before dispersing and the flats can be applied after.

But this implies a whole new set of reference files, which will need to go into CRDS, which means it will be a while before they are ready. Perhaps in the short term, we apply the flat field after dispersing (i.e. in obs_generator.py), and put in place a dummy "transmission file" that is applied before dispersing. We can use the dummy file until true transmission files have been produced. It seems like it shouldn't be hard to pull out the occulters from a NIRISS flat and make a transmission file that is mostly correct.

What do you think @KevinVolkSTScI ?

bhilbert4, 2020-07-16:
I consider this very high priority and would like to get it solved quickly.

bhilbert4, 2020-07-16:
Tagging @NorPirzkal , to keep him in the loop.

KevinVolkSTScI, 2020-07-16:
We can take the flat field out of the seed image for NIRISS imaging.  Whilst it is useful to see the effect of the occulting spots and the other pick-off mirror features, this is not essential.  One can always multiply the seed image by the flat outside Mirage for comparison with the pipeline ramp slope output files as needed.  As long as the flat is applied before the ramp is simulated it is OK.  But we cannot do that for the dispersed image case.

Chris Willott and Swara Ravindranth want to produce a new reference file for the POM transmission separately from the flat field.  This is needed for the order overlap.  The main problem is that this needs to be the size of the POM itself, and we have no information for the area outside the detector region.  A secondary problem is that the POM transmission is known to be wavelength dependent.  These new POM transmission files can be used in place of the flat field in the WFSS case in making the scene image before it is dispersed.  How soon this might be ready to go to CRDS is not clear, but it could be make available for Mirage if not in CRDS fairly quickly.  Chris produced flats with the POM features removed (and the ones already in CRDS made by me also do this in a similar manner).  Those will be submitted to CRDS fairly soon.

NorPirzkal, 2020-07-16:
Hi

After talking with Chris and Swara this afternoon, we discussed an implementation that is filter dependent. This will account for a large fraction of the wavelength dependences of most observations.
While we might or might know the size of the POM, we already have estimates of that for NIRCAM, and I assume there are for NIRISS also. The pipeline will need these files also when computing contamination so I think we should start implementing simple versions as a first step and for use with Mirage. These can be improved and delivered to crds when we can but the pipeline people need to understand that these are needed for the extraction process.
I would really encourage everyone to not refer to these as flatfields as this really confuses things. These are input transmission maps.

N

Dr. Nor Pirzkal
ESA/AURA Astronomer Level III
Space Telescope Science Institute
Baltimore, MD, USA

bhilbert4, 2020-07-16:
Right. There's always a lot of overhead getting things into CRDS, so I don't expect to be able to have Mirage get transmission files from there for quite a while. But in the meantime, would a placeholder transmission file be acceptable? The way Mirage works at the moment, by applying the flat, which is only 2048x2048, it is essentially assuming that transmission outside the detector is 100%. So switching to a dummy transmission file with the transmission in the POM set to 1.0, and the occulter pixels pulled from the current flat, would be the same, right? If that makes sense, then Mirage could be updated now to get the flat fielding correct for all modes (i.e. when the seed image, dispersed or not, is translated into a ramp), while having the basic mechanism in place to apply a transmission file to the seed image prior to dispersing.

NorPirzkal, 2020-07-16:
I would suggest implementing things in a more general way with the transmission map being x by x pixels (say 2800x2800 but any size really) and then have two header keywords that point to the coordinate of the bottom left corner of the actual detector in that image. Mirage should be able to use that information to properly apply that transmission map to the seed image.

KevinVolkSTScI, 2020-07-16:
It sounds like we should assume that there is a filter-dependent transmission file for the WFSS mode that will live in the $MIRAGE_DATA directory somewhere and be a parameter in the .yaml file, and then fetch it from CRDS later when these are available there.

bhilbert4, 2020-07-16:
Yes, I think that makes sense, although my hope is that in the short term those files live in the GRISM_NIRCAM and GRISM_NIRISS repos. :)

NorPirzkal, 2020-07-16:
They can be….

chriswillott, 2020-07-17:
I agree with the implementation you propose above. I have just produced filter-dependent, oversized, POM transmission masks for NIRISS. I will confer with @KevinVolkSTScI about these since I know he has has also done some work on determining the out-of-field extent of the POM. I used header keywords OFFSETX and OFFSETY for the offsets in pixels between the lower-left corner of the transmission file and the detector, allowing these files to have a variable size as Nor suggests above. 

As Kevin points out above, the NIRISS grism flat fields in CRDS do not contain the occulting spots - they have been patched over. I have provided Kevin with new versions of the flats to go into CRDS soon, but these are only a minor change compared to those in use now.

bhilbert4, 2020-07-28:
@chriswillott @KevinVolkSTScI I have the code changes more or less complete to support this. Do you have the transmission mask files for NIRISS ready? For the time being I think we're going to host them Nor's GRISM_NIRISS repository.

bhilbert4, 2020-08-07:
I worked a little more on this yesterday after discussions with Nor. The way the code is set up now, the size of the transmission image is what defines the size of the seed image and segmentation map. For imaging modes, Mirage will read in the transmission image and crop it down to the size of the detector using the NOMXSTRT, NOMYSTRT header keywords (which give the coordinates of the lower-left pixel of the detector within the larger transmission image. This image is multiplied into the seed image as one of the final steps in catalog_seed_image(). 

For WFSS modes, the transmission image is not cropped. The seed image and segmentation map are made the same size as the transmission image. In this case, Mirage does not multiply the transmission image into the seed image though. Instead, when the seed image is sent to the disperser, the disperser takes the transmission image and uses it while dispersing.

I haven't put in the changes yet, but for the "pom" mode for NIRISS, my plan is to have it create the seed image using the full-sized transmission image (like WFSS mode), and to multiply it into the seed image before saving (like imaging modes).

chriswillott, 2020-08-12:
I'm not sure what you mean by "pom" mode for NIRISS. The way the transmission files should be used for NIRISS are:

Imaging mode: do not use the transmission file at all. All the features in the transmission file are already included in the imaging flat field files so they get applied when doing the flat-field step in Mirage.

WFSS mode: Exactly as you write above in the "WFSS modes" paragraph:  the transmission image is not cropped. The seed image and segmentation map are made the same size as the transmission image. In this case, Mirage does not multiply the transmission image into the seed image though. Instead, when the seed image is sent to the disperser, the disperser takes the transmission image and uses it while dispersing.

For NIRISS WFSS mode, you could multiply the transmission into the seed image before saving instead of doing it at the disperser step, but I think it is easiest to keep the same system for NIRISS and NIRCam, especially since the code would have to reside in two different repos: mirage and NIRCAM_Gsim.

NorPirzkal, 2020-08-12:
The way things are currently coded in the WFSS module for NIRCAM, the POM fits file contains the mask in the first extension. Tables containing the transmission of optical any occulter to account for is stored in addition extensions. Any number > 10000 in the POM image is taken to indicate that an additional transmission needs to be applied to the input spectra before they are dispersed (whether these are actual input spectra or spectra internally derived from the given imaging data).
So for NIRISS, what we need are a POM image which is large enough to contain everything that might get dispersed. The bottom left corner of the actual detector should be stored in the keywords NOMXSTRT,NOMXEND,NOMYSTRT, and NOMYEND, which are the same keywords used by Mirage in the seed images it generates. You can set the POM values to 0 or 1, if you have regions containing pixels where you would like an additional sensitivity to be applied by the WFSS module, use any value greater than 10000 instead (10001 for NDF, 10002 for COM substrate for NIRCAM for example). You can send me the transmission file(s) and the POM image and I can take care of generating the files in the format needed for the WFSS module.

KevinVolkSTScI, 2020-08-12:
In Mirage you can ask for a NIRISS POM mode imaging simulation, whereupon the code produces a regular imaging simulation covering the entire POM in the seed image.  In such a case one would use the new transmission file and turn off the flat field reference file.  Then one can for example convolve the seed image with the SOSS "point spread function" to get the expected SOSS scene with all the sources and the proper background shape.

bhilbert4, 2020-08-12:
One of the main drivers behind this work was to separate the transmission image from the flat field, primarily to get the WFSS simulations correct. For imaging mode, I suppose it doesn't matter if the NIRISS occulters are in the flat field file or the new transmission image though. If you want to keep the occulters in the imaging mode flats, then I could turn off transmission images for imaging mode completely. Let me know if that is what you prefer. I'll keep the use of the transmission file for NIRISS POM mode. 

With these changes, for all modes, the flat field is not applied until observation generation step, which means that it will not be applied in the seed image as it comes out of catalog_seed_image().  Should I turn off the application of the flat field for POM mode, now that it is done in the observation generation step?

KevinVolkSTScI, 2020-08-12:
I would say yes, turn off the flat field step in the POM mode case.

NorPirzkal, 2020-08-12:
I would say no. First because the flat field should be applied logically where it belongs and two because this would keep nircam and niriss in check. This will also avoid further confusion leading to any double flat fielding as it has been done with niriss data.

bhilbert4, 2020-08-12:
I think there was a mix-up with terminology here. In Mirage's POM mode, the transmission image (aka "POM mask") will still be applied. Kevin was talking about turning off the flat fielding in the observation generation step when running a POM mode simulation.

NorPirzkal, 2020-08-12:
good to know.

closed 2020-10-09 by bhilbert4