spacetelescope/mirage #526: NIRISS PSFs in gridded_psf_library should be normalized to 1

dthatte, 2020-07-23:
The scaling factor for magnitude to count rate conversion is accounted for in utils.py, lines 721 to 729 and values in the photometry file for NRM and regular imaging. Therefore the PSFs in 
the gridded_psf_library should be normalized to 1.

```
# For NIRISS AMI mode, the count rate values calculated need to be
# scaled by a factor 0.15/0.84 = 0.17857. The 0.15 value is the
# throughput of the NRM, while the 0.84 value is the throughput of the
# imaging CLEARP element that is in place in the pupil wheel for the
# normal imaging observations.
if observation_mode in ['ami', 'AMI']:
   count_scale = 0.15 / 0.84
else:
   count_scale=1.
```

For example niriss_nis_f480m_mask_nrm_fovp301_samp5_npsf1_predicted_realization0_nis.fits is normalized to 0.15057966144945326 and niriss_nis_f480m_clearp_fovp301_samp5_npsf1_predicted_realization0_nis.fits is normalized to 0.8357325488232911 .

When I used the above NRM PSF the count rate in the seed image was too low because the NRM throughput was accounted for twice, once because of the scaling factor and the second time because of normalization to ~0.15.

I compared the count rate with a calculation for the same source and exposure parameters in the JWST ETC. The count rate in the central pixel was very close to the count rate in the JWST ETC (displayed in the attached screenshot) after normalizing the NRM PSF to 1.

[image]
KevinVolkSTScI suggested to check if NIRCam PSFs for the specialized phase masks have the right count rate with the current set-up.

mperrin, 2020-07-24:
Hmm, let’s see. We’ve previously checked the normalization for consistency with photutils and mirage; see https://github.com/spacetelescope/webbpsf/issues/302. But maybe we missed something. Are you saying specifically about the AMI PSFs only but not regular imaging? 

The convention for normalization adopted between WebbPSF and Pandeia is that Pandeia handles everything that is not spatially dependent on some shapes mask put in an optical system and doesn’t affect the shape of the PSF (so, mirror reflectivities and filters and defector coatings and so on), and WebbPSF handles everything that does affect the shape of a PSF (so, coronagraph masks and AMI masks and spectrograph slit losses). A consistent implementation in MIRAGE would say for instance that a star has some given count rate in F480M (independent of any masks in the beam) and then that gets multiplied by a PSF that reduces the intensity by 0.15 if it’s F480M plus the AMI mask. If MIRAGE is separately applying another factor of 0.15 itself to the AMI count rates then yeah it does sound like there is some double counting.

bhilbert4, 2020-07-24:
That scaling is something added by @KevinVolkSTScI in #161. Maybe at that point we were using PSFs that did not include the effects of the AMI mask in the scaling? I believe that was before we switched to the gridded PSF libraries. Anyway, since the PSF from the gridded library looks to be scaled correctly, it makes sense to me that we can remove the block of code above from Mirage.

KevinVolkSTScI, 2020-07-24:
Before the gridded PSFs were added the PSF library images were all normalized to 1 when they were read in because the photometry values to conveert from magnitude to count rate include the full throughput including the aperture effects that WebbPSF has in the PSF normalization.  The count rate values for the AMI case are scaled by 0.15/0.84 to allow for the differential throughput compared to normal imaging.  As it stands now with the PSFs not being renormalized to 1.0 the long wavelength NIRISS filter counts are too low by a factor of 0.84 and the NRM ones are too low by a factor of 0.15.

Just removing the scaling factor 0.15/0.84 for the AMI mode case will leave both the NRM and normal imaging values for the longer wavelength filters too low by a factor of 0.84.  Hence we need to increase the count rates by 1./0.84 for the NIRISS long wavelength filters with or without the NRM, or normalize the PSFs back to 1.0 in all cases.  Either one will work.

mperrin, 2020-07-24:
I agree Kevin, that makes sense to me. I would vote for the first option, because I think it is better in the long run to keep the same PSF normalization conventions between Pandeia and MIRAGE. But I agree either would work and both approaches are acceptable.

bhilbert4, 2020-07-24:
See PR #527. I had to play a bit with function inputs in order to be able to differentiate based on filter, but everything worked in my tests. See the changes in utils.py. @KevinVolkSTScI does that change look correct?

PR #527 (2020-07-27): Update countrate scaling for NIRISS AMI and filter wheel filters
This PR updates scaling applied when converting magnitudes to count rates for NIRISS simulations. The previous scaling values were applicable when the PSF from the PSF library was normalized to 1.0, which is no longer the case. With the gridded PSF libraries in place, the throughput factor of 0.15 for the NRM is incorporated into the webbpsf output, so we remove that from the scaling. Also, the PSFs from webbpsf include the throughput factor of 0.84 for the CLEARP optical element. This 0.84 factor is also baked into the photflam/ [... cut]
closed 2020-07-27 by KevinVolkSTScI