spacetelescope/mirage #533: File splitting limit is not sufficient for engineering mode and commissioning

kjbrooks, 2020-08-05:
When using mirage for NIRCam simulations, I ran into an error that @bhilbert4 was able to track to down to be connected to file splitting based on NINT/Groups. I am copying his email below:

> I implemented file segmenting in Mirage when I added TSO mode last summer. This will look at your exposure and if the data volume is over some limit, Mirage will split it up into multiple output files. I tried my best to mimic the splitting that the jwst pipeline is going to do. The problem is that I only really paid attention to file splitting for TSO mode where there will be tons of integrations. I didn’t think about it for regular imaging mode, because GO observations are required (I think) to have no more than 10 groups, which is under the file splitting limit. ... in engineering mode, and in commissioning, there are plenty of exposures that are over the file splitting limit.

bhilbert4, 2020-08-05:
This should be fixed now, via #506. Let me know if you are still getting file splitting errors.

kjbrooks, 2020-08-05:
oh great! apologies for missing that. Closing this issue.

closed 2020-08-05 by kjbrooks