spacetelescope/mirage #534: FGS catalog creation not quite robust

kjbrooks, 2020-08-05:
The currently implemented FGS catalog creation does not appear to be robust. The following are the two known issues with `mirage.catalogs.create_catalog.get_all_catalogs` for FGS:

- Bug: `mirage.catalogs.create_catalog.get_all_catalogs` documentation says to use "Guider" for FGS but, the code actually expects "FGS"
- According to @KevinVolkSTScI (from an email I received from him): "The code in Mirage interpolates from the input catalogue photon weighted mean flux density values as a function of pivot wavelength to the pivot wavelength of the filter under consideration.  This is the best general method because if there is extinction then the magnitudes are affected but interpolation should generally give a reasonable result.  However, this actually only works if the JWST filter is of a similar wavelength width as the input 2MASS/WISE/GAIA filters.  For the Guider this is not going to work because the response is so wide that it is dominated by the short wavelength part of the stellar spectrum but the pivot wavelength is around 2.3 microns.  So in essence the “proper” magnitude is roughly similar to a V or NIRISS F090W filter magnitude, while the interpolation will give you something like the K magnitude.  For cooler objects, the magnitude will be too bright when the Guider filters are used."

For the second point above, there are two alternatives to figuring out the FGS countrates/magnitudes: using `jwst-magnitude-conversion` or `jwst-fgs-countrate`. These two modules will give FGS count rates/magnitude but they arrive at these values in different ways:

- [`jwst-magnitude-conversion`](https://github.com/spacetelescope/jwst_magnitude_conversion) uses stellar atmosphere model spectra to simulate the different magnitudes

- [`jwst-fgs-countrate`](https://github.com/spacetelescope/jwst-fgs-countrate) uses stellar energy distributions that are derived from the observed GSC2, 2MASS, and SDSS magnitudes to compute the FGS count rate or magnitude

Both tools use the characteristics of the science instrument detectors (e.g., throughput, gain) to compute the instrument magnitudes.

Since I have some experience with the `jwst-fgs-countrate` tool, I have written a way to use it to create FGS catalogs for the use of the JWST OTE team for OTE commissioning. It can be found here: https://grit.stsci.edu/wfsc/tools/-/blob/master/mirage-ote-utils/mirage_ote_utils.py#L475 

I am not sure what is in the best interest of mirage as a whole which is why I made this issue and didn't just submit a PR. There is the option to use the bit of code I have written up when trying to create an FGS catalog if it is believed to be complete enough.

bhilbert4, 2020-08-06:
I don't see any reason not to include your code in Mirage and use it when creating catalogs for FGS. I think it would make sense to add `jwst-fgs-countrate` as a Mirage dependency since it is doing the calculations correctly while the current code in Mirage does not. 

With this change, does the standard method for converting from magnitudes to countrate still work for FGS? https://github.com/spacetelescope/mirage/blob/master/mirage/utils/utils.py#L686

On a related note, a while back we started to explore, with the FGS team, supporting catalogs that contain count rates rather than magnitudes. (See #355) This work was abandoned when Johannes left, and we didn't hear back from the FGS team about whether this would be useful/desirable or not. Do you think it would be useful to resurrect this work?

kjbrooks, 2020-09-29:
I am trying to squeeze some work in on this between rehearsals so I apologize for the delay. 

I have been looking more into what the `fgscountrate` package is doing and right now its primary goal is to calculate magnitudes and count rates from catalog bandpasses for individual objects. It does this simultaneously. It looks like Johannes had requested a module that converts mag to count rate (https://github.com/spacetelescope/jwst-fgs-countrate/issues/3). In discussion with @shanosborne, it sounds like she has the math for this conversion but hasn't been able to get it into the code yet. It looks to me like the calculation she has is in the form of the vegamag magnitude system here https://github.com/spacetelescope/mirage/blob/master/mirage/utils/utils.py#L686. So, to answer your question, the standard method for converting from magnitudes to count rates will work, with the correct `vegamag_zeropoint`, though with the use of the `fgscountrate` package creating a catalog, you wouldn't need to use it. 

I am not sure if there is a current need for catalogs that contain count rates rather than magnitudes. I have not heard anything to this extent and cannot think of an instance when that would be necessary, though I think that once this work is complete, I can't imagine it would be too difficult to implement (I say that now...).

bhilbert4, 2020-09-30:
So just to get an idea for the workflow, and make sure I have things right, the user would create a Mirage catalog with `fgs1_magnitude` and/or `fgs2_magnitude` columns. The calculation of those magnitude values would be done presumably with `jwst-fgs-countrate` or `jwst-magnitude-conversion`. When Mirage is run, it would then call `jwst-fgs-countrate` (once the proper code is added) to do the conversion from magnitudes to count rate?

kjbrooks, 2020-09-30:
That's a good question. This is how I am imagining it: the user will create the FGS catalog in mirage (while mirage will outsource the work to get magnitudes to `jwst-fgs-countrate`). Right now with `jwst-fgs-countrate`, the count rate comes for free with the magnitude, so if we can find a way to include that in the catalog, that could save time. However if we don't want to do that (or can't or shouldn't) then I think you can simply call `magnitude_to_countrate` in `utils.py` with the proper parameters passed through (I have to look into how the count rates get called but it seems like this will be straightforward). I think ultimately we don't need `jwst-fgs-countrate` to have the magnitude to count rate conversion since the one in mirage will work fine. Does that seem reasonable to you? I don't fully understand the inner workings of mirage so it is completely possible that this isn't actually feasible.