spacetelescope/mirage #548: Add optional jitter to simulations

bhilbert4, 2020-08-27:
Allow users to add jitter to their simulations if they want to.

JWST jitter RMS is 7mas if I recall, but I'm not sure what timescale that is on. Should we add jitter by simply tweaking the pointing of each exposure? Or do we need to tweak the locations of targets within an exposure/integration?

If the latter, then we should probably add a 'jitter' entry to the yaml files, with a value in arcseconds representing the jitter to use.

mperrin, 2020-08-27:
WebbPSF includes the 7 mas jitter, as a convolution term on top of the diffractive PSF. The time scale for this is -much- faster than any JWST exposures except a few very specialized time series cases; multi hertz. 

For the early WFSC activities we handle this in the segment PSF generation. The same approach naturally allows putting arbitrary jitter in in-focus PSFs too, if that’s of interest

bhilbert4, 2020-08-27:
Interesting! So the gridded PSF libraries that Mirage uses include the default jitter (since I didn't modify any jitter-related options when calling Webbpsf to create them)?  That's good to know. The TSO folks have been wondering about adding jitter to their datasets, but it sounds like the jitter is already in there. I think the idea of changing the magnitude of the jitter is lower priority than simply having jitter effects.

mperrin, 2020-08-27:
> So the gridded PSF libraries that Mirage uses include the default jitter (since I didn't modify any jitter-related options when calling Webbpsf to create them)?

Yes. The exact same 7 mas Gaussian convolution is applied to all PSFs by default. Take a look at the FITS headers and WebbPSF saves some info about this, for instance what the Strehl ratio reduction from the jitter is. Typically a small change.

TSO may also care about small differences in time-averaged pointing from one exposure to the next. Slow drifts on longer time scales. By design the FGS+FSM+ACS in closed loop operation should keep any such drits small under the required thresholds, but if the TSO folks want to evaluate the precise effects of small nonzero differences within the budgets that would take more effort. 

The 7 mas is a top-level requirement, which combines multiple budget terms on different timescales (faster than fine guide bandwidth, in the fine guide control bandwidth, and very slow drifts over hour+ timescales from motions not tracked in the guide loop at all such as drifts of one SI relative to another). 

Before doing any more work on this, I would think we/you would want to get a clear sense of what particular details the TSO folks are interested in, on what timescales for observations. Probably also should involve some of the FGS & LOS experts in our group in that too.