spacetelescope/mirage #553: dark files missing in yaml file

aceilers, 2020-09-08:
Hi, 

I'm trying to mimic NIRCam WFSS data using mirage for the GTO program 1243. However, it crashes because both "dark" and "linearized_darkfile" are set to None in the automatically created .yaml file (when running yaml_generator.py). I downloaded all the reference files, can I just set the linearized dark files manually in the .yaml file? If so, how do I know which ones to use? Or is there some prior mistake due to which there are no filenames for the dark files in the .yaml file?

Thanks for your help!
Christina

bhilbert4, 2020-09-10:
Hi Christina,

Sorry for the delay. I think this is an issue with yaml_generator.py. It should have populated those dark file names automatically. Do you have the MIRAGE_DATA environment variable set, and the linearized dark files (for the A5 detector in this example) sitting in MIRAGE_DATA/nircam/darks/linearized/A5/ and/or the raw darks in MIRAGE_DATA/nircam/darks/linearized/A5/ ? The yaml_generator looks into those directories to get a list of the available darks to be used. There was a bug a few months ago where the yaml_generator required both the linearized and raw darks to be present in the collection of reference files. I fixed that issue, but the fix is not present in the released version of Mirage on Pypi.

You certainly can manually set the "dark" or "linearized_dark" value in each yaml file if you would like, or if you don't feel like dealing with whatever is going wrong with the yaml_generator. As long as you use a dark for the proper detector specified in the yaml file, it doesn't matter which of the 5 darks you put in each yaml file. I would recommend mixing them up (i.e. don't use the same linearized dark for all exposures) just to make things a little more realistic.

Bryan

cnaw, 2021-01-14:
Hi Bryan,

I am finding a very similar (if not identical issue). I am reading an APT output (via XML and pointing) and feeding it to  yaml\_generator.SimInput\.  
In my script when it gets to  yam.create_inputs()  it does write some output but then crashes because no darks are found (I had to hack yaml.generator to print out some of the lines below):

--------------------------------------------------------------------------------------------------
will run yam.create_inputs()
csv exposure list written to /home/cnaw/python/mirage_test/Observation_table_for_1073.csv
657  nircam A1
901  {'nircam': {'A1': [], 'A2': [], 'A3': [], 'A4': [], 'A5': [], 'B1': [], 'B2': [], 'B3': [], 'B4': [], 'B5': []}, 'niriss': {'NIS': []}, 'fgs': {'G1': [], 'G2': []}, 'miri': {'MIR': 'none'}, 'nirspec': {'NRS': 'none'}}
903  nircam A1 [] 0
Traceback (most recent call last):
  File "test2_use_apt.py", line 41, in <module>
    yam.create_inputs()
  File "/home/cnaw/anaconda3/envs/mirage/lib/python3.7/site-packages/mirage/yaml/yaml_generator.py", line 658, in create_inputs
    darks.append(self.get_dark(instrument, det))
  File "/home/cnaw/anaconda3/envs/mirage/lib/python3.7/site-packages/mirage/yaml/yaml_generator.py", line 907, in get_dark
    rand_index = np.random.randint(0, len(files) - 1)
  File "mtrand.pyx", line 992, in mtrand.RandomState.randint
ValueError: Range cannot be empty (low >= high) unless no samples are taken
(mirage) [cnaw@orange mirage_test]$ printenv MIRAGE_DATA
/data1/mirage/mirage_data
------------------------------------------------------------------------------------------
The code crashes because in the random number test there are no files
I tried using reffile_overrides to define the darks but that did not work either (probably some syntax issue on my part).
Do you have any suggestion on how I can fix this ? 
I also notice that for the darks ALONG and BLONG are used instead of A5 and B5. Do the scripts take that into account?
[279]cnaw@orange:/data1/mirage/mirage_data/nircam/darks/linearized% ls
A1/  A2/  A3/  A4/  ALONG/  B1/  B2/  B3/  B4/  BLONG/
Thanks!
Christopher

cnaw, 2021-01-14:
Hi Bryan,
This problem seems to go away if I install the development version. Sorry for the false alarm!
Christopher

closed 2021-05-19 by bhilbert4