spacetelescope/mirage #562: Option to fetch fewer darks for simple testing

gbrammer, 2020-09-29:
Feature request: provide an option in ``downloader`` to fetch only a limited set of dark reference files rather than forcing the user to download  hundreds of GB of reference files just to get a test up and running!

Thanks!

bhilbert4, 2020-09-30:
Are you thinking of an option that retrieves just a single dark file for each detector? Or have the user specify a detector and download a single dark for only that detector? While developing, my thought was that users will generally want to be able to create simulations of all detectors, or at least all detectors of a given module. 

Also, the raw darks are smaller, and faster to download, but that time savings is pretty much lost by then having to run the calibration pipeline on those files.

gbrammer, 2020-10-01:
Yes, I think the main thing I had in mind was single darks for each detector.  That already decreases the download size by a factor of 10.

PR #579 (2020-12-07): Add option for downloading a single dark for each detector
This PR adds an option to the reference file downloader to download only a single dark current file per detector. This is intended for those who want a quick start of sorts, to test out Mirage, without downloading all of the reference files. This is done via the added `single_dark` keyword in the downloader script.

Resolves #562
closed 2020-12-07 by bhilbert4