spacetelescope/mirage #566: Invalid readout pattern value appears in obs params

coljac, 2020-10-08:
I am attempting a NIRISS WFS simulation using my own SED file and a galaxy catalog. I can produce the seed image but then run into the following exceptions:

```
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-76-0283a71a434d> in <module>
      9                                create_continuum_seds=False # Also unclear
     10                               )
---> 11     m.create()

~/.conda/envs/mirage/lib/python3.6/site-packages/mirage/wfss_simulator.py in create(self)
    342             d = dark_prep.DarkPrep(offline=self.offline)
    343             d.paramfile = self.wfss_yaml
--> 344             d.prepare()
    345             obslindark = d.prepDark
    346         else:

~/.conda/envs/mirage/lib/python3.6/site-packages/mirage/dark/dark_prep.py in prepare(self)
    922             nints, ngroups, yd, xd = final_dark.shape
--> 923             h0.header['READPATT'] = self.params['Readout']['readpatt'].upper()
    924             h0.header['NINTS'] = nints
    925             h0.header['NGROUPS'] = ngroups

AttributeError: 'numpy.int64' object has no attribute 'upper'
```
And:
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-77-baed8b0c9604> in <module>
      9                                create_continuum_seds=False # Also unclear
     10                               )
---> 11     m.create()

~/.conda/envs/mirage/lib/python3.6/site-packages/mirage/wfss_simulator.py in create(self)
    375         #obs.paramfile = y.outname
    376         obs.paramfile = self.wfss_yaml
--> 377         obs.create()
    378 
    379     def param_checks(self):

~/.conda/envs/mirage/lib/python3.6/site-packages/mirage/ramp_generator/obs_generator.py in create(self)
   1089         self.params = utils.get_subarray_info(self.params, self.subdict)
   1090 
-> 1091         self.check_params()
   1092 
   1093         # Read in cosmic ray library files if

~/.conda/envs/mirage/lib/python3.6/site-packages/mirage/ramp_generator/obs_generator.py in check_params(self)
    691         # Check that readout patterns of input dark and requested output
    692         # are compatible
--> 693         self.readpattern_compatible()
    694 
    695         # Make sure ngroup and nint are integers

~/.conda/envs/mirage/lib/python3.6/site-packages/mirage/ramp_generator/obs_generator.py in readpattern_compatible(self)
   2560                               "pattern of {}. The readout pattern of the dark "
   2561                               "must be RAPID, NISRAPID, FGSRAPID, or match the requested output "
-> 2562                               "readout pattern.".format(darkpatt, self.params['Readout']['readpatt'])))
   2563 
   2564     def ref_check(self, rele):
```
Looking at the params it is complaining about, I see:
```

{'Inst': {'instrument': 'NIRISS', 'mode': 'wfss', 'use_JWST_pipeline': True}, 'Readout': {'readpatt': 27, 'ngroup': 40, 'nint': 2, 'namp': 4, 'resets_bet_ints': 1, 'array_name': 'NIS_CEN', 'filter': 'GR150R', 'pupil': 'F150W', 'nframe': 4, 'nskip': 0},....}
```
`readpatt` is 27. In the YAML (attached), it's "NIS".

Could there be a bug here, which has popped up due to having too many groups in my input APT XML? 
I see in [obs_generator.py](https://github.com/spacetelescope/mirage/blob/b6aa07bf92154805cbc2c6b96805a8786b3c95c9/mirage/ramp_generator/obs_generator.py#L747-L751) it's testing for `self.params['Readout']['ngroup']` then if it's too many, setting ` self.params['Readout']['readpatt'] = maxgroups `. I'm only guessing, but would ` self.params['Readout']['ngroup'] = maxgroups ` make more sense, and be the cause of my issue?

[yaml.zip](https://github.com/spacetelescope/mirage/files/5345628/yaml.zip)

bhilbert4, 2020-10-09:
Thanks for bringing this up. You're right. `maxgroups` is being put into the wrong parameter. It should go into `self.params['Readout']['ngroup']`. I turned off the group number checking in `catalog_seed_image.py` a while ago. I should have turned it off in `obs_generator.py` at the same time. APT enforces maximum numbers of groups, but there's no real reason for Mirage to enforce these rules as well.

closed 2020-10-09 by bhilbert4

bhilbert4, 2020-10-09:
I've pushed a fix to the master branch that should solve this problem. I opted to remove the ngroups check altogether and rely on APT to provide warnings about requesting too many groups.

PR #567 (2020-10-09): Remove the check/adjustment to the max number of groups.
This PR removes the check on the number of groups in the input yaml file. Previously if the requested number of groups was higher than the given limit (which came from APT), then Mirage would lower the number of groups in the simulation to match the maximum allowed. This logic was removed from `catalog_seed_generator.py` a while ago, but remained in `dark_prep.py` and `obs_generator.py`. Since Mirage inputs mostly come from APT anyway, I don't think Mirage needs to enforce a maximum groups rule. This is something better left to APT itsel [... cut]