spacetelescope/mirage #592: Add optical ghosts to NIRISS data

bhilbert4, 2020-12-04:
@mtakahiro has come up with a method to add optical ghosts that correspond to input point sources to NIRISS simulations. The ghosts' locations and brightnesses depend on the filter being used (and possibly on location on the detector). 

Incorporating this into Mirage shouldn't be too difficult. Based on his notebook, the algorithm would be something like:

Read in the point source catalog and calculate locations on the detector (in x,y), as well as count rate. Provide this information to a new function that calculates the ghost location and count rate. Select the stamp image to use for the ghosts based on filter (and maybe location). Update (or create) and extended source catalog that contains the ghost entries, including the appropriate stamp image file. Since extended sources are handled after point source, it should be possible to do this update on the fly.

For WFSS data, the ghosts should have the same SED as the sources that cause them. In this case, there are several options:

- If the SED is derived from one or more catalog magnitudes through various filters, then we need to calculate the ghost's magnitude in these filters as well (based on the real source's magnitudes).
- If the user provides an hdf5 file with SEDs, then that catalog file will need to be updated as well, copying the SED from the original source
- If the user provides a dictionary containing SEDs, then the dictionary will have to be updated in the same way as the hdf5 file above.

bhilbert4, 2020-12-07:
Here's a complicating factor: Often the same catalog is used for multiple exposures. If Mirage were to update the catalogs to contain the ghost sources, then you would only want to do it once. Otherwise new ghost objects would be added every time Mirage is run, as it creates ghosts of ghosts. Options to get around this:

-  The catalogs are updated and saved and then in subsequent calls to Mirage, there needs to be a way to tell Mirage to skip the addition of ghosts. This seems doable, and in either case, adding a keyword to the yaml file to turn ghost production on and off seems like something to add. The problem here would be populating the keyword value. Mirage wouldn't be able to do it via the yaml_generator without forcing the user into running exposure N first, followed by the others.

Advantages: New ghosts are permanently saved in catalog files. No need to repeat calculations. Run as part of the standard call to Mirage.
Disadvantages: Confusing for the user. Good chance that extra, unreal sources could be added to the catalog.

- Make the ghost production a catalog tool outside of catalog_seed_image(). In this case, the user creates source catalogs, runs this new tool, which modifies the catalogs, and then runs the yaml_generator in order to add the catalogs to all exposures. In this case, the seed creation code (catalog_seed_image, imaging_simulator) would know nothing about ghosts. It would just be working from catalogs that already contain all the ghosts.

Advantages: New ghosts are permanently saved in catalog files. No need to repeat calculations. Run as part of the standard call to Mirage.
Disavantages: This would be one more step that a user would have to manually run before calling Mirage. (Perhaps it could be folded into the calls to catalog_generator somehow?

- Keep the ghost source creation inside of catalog_seed_image(), but keep the new ghost sources local. (e.g. create a temporary copy of the input extended source catalog and add the ghosts, and update only the table of sources read in from the hdf5 file, but do not update the file itself.) 

Advantages: Run as part of the standard call to Mirage.
Disadvantages:  Calculations relating to the ghosts would have to be repeated each time Mirage is run.

I'm leaning towards option number 2. @mtakahiro, any thoughts?

mperrin, 2020-12-07:
Isn’t there also another complicating factor:  different observations using different position angles/orientations? Presumably the offset of a ghost is fixed in the instrument reference frame, yes? Therefore any modified catalog should not be reused for any observation at a different orientation. And then furthermore I would guess it’s a function of wavelength filter? 

I’m not an expert in NIRISS by any means, just speculating here and thinking out loud, but it seems hard to me to reuse catalogs in general once you start adding in instrumental artifacts like this.

mtakahiro, 2020-12-07:
Thanks, @bhilbert4 !

I was hoping something that users can simply provide yaml files with ghost flag (True or False), then if True this new functionality goes through 1.adding ghost based on the real source catalog and detector position (from the provided yaml file), 2.update (or create a new) yaml file to have the newly created, source+ghost catalog, 3.run either of the imaging or WFSS simulators in one row.

As @mperrin suggested, I'd think that having multiple catalogs (for each yaml) written in the disk is unavoidable. Since the ghost position is calculated in the pixel coordinate (not sky position), the new catalog created in the step 1 should be provided in position_pixels too, whereas I imagine that the original source catalog by user can be either of pixel or sky coordinates.
I understand that this would produce a bunch of catalogs and possibly yaml files, but should be okay as user may want to know which of the sources in the output images are ghosts. 

So I guess what I'm saying here is close to the option 2, but with the catalog generation process also included in Mirage. Maybe I'm missing something important, especially wrt minimizing calculation time, or else. If that's the case, please let me know!

bhilbert4, 2020-12-07:
Ah yes, that's true. That argues for option 3 then, where ghost sources are created within the call to catalog_seed_image, and the original input source catalogs are left untouched.

closed 2021-01-14 by bhilbert4