spacetelescope/mirage #604: In NIRISS WFSS or POM modes the direct images are limited to the normal detector area

KevinVolkSTScI, 2021-01-19:
In NIRISS POM mode or WFSS mode the direct image simulation should be 2322x2322 pixels.  It used to be that way, say 1 year ago.  Currently the seed images for these cases have the central 2048x2048 pixels of the seed image populated but the surrounding 137 pixel border where the POM is present but the images do not fall onto the detector is all zero.  I had thought that this was corrected, but the Mirage version as of a few days ago shows this problem.  My old Mirage version of 13 February 2020 does not have this issue and the images are fully populated.

bhilbert4, 2021-01-19:
I haven't run a POM simulation in a while, but I just ran a NIRISS WFSS simulation this morning and my seed image was 2429x2429, with the source properly extending all the way to the edges. The 2429 size is dictated by the POM transmission files that were added recently. I'll run a POM mode case soon and see if I get the same result you do.

KevinVolkSTScI, 2021-01-20:
OK I tried the POM example again in a new window/session and it worked as expected this time.  It is possible that conda is messing up the version of Mirage that is being used.  I am re-running the WFSS test case now to see if it also works as expected with the new window.  But the log says that there 1222 sources in or near the area and of these 843 are being dispersed which looks like it is not doing the full area.  It will be a while before the model is finished.  The size of the output seed image in the POM case is 2322x2322 pixels.  That does not match what you said in your comment.

bhilbert4, 2021-01-20:
I just checked the code to remind myself. POM mode is treated like an imaging exposure, so the transmission file is not used, so the array is being expanded by the same factor as Mirage has been using. So that's why it remains 2322x2322. Only WFSS observations will have seed image sizes that are defined by the transmission files. 

As for the number of sources in the log, the first count, of the number of sources in or near the detector, still includes sources that are outside the detector. That's meant to give a rough idea of the number of sources. It's primarily used to throw out sources that are far from the detector and make the source catalog smaller, in order to save computation time later. The count of the number of sources being dispersed should be correct. It should include all sources on the detector or in the buffer area around the detector, whose traces may extend onto the detector.

KevinVolkSTScI, 2021-01-22:
Bryan, I just ran another test with Mirage version 1.3.3 and it crops out the seed images in the output files when I run an F090W grism exposure calculation.  I am not sure that I am getting consistent results from one run to the next.  I thought I ran an imaging test for POM mode in the current version and got the 2322x2322 pixel field but I re-ran this and got only the central area being populated.  So I am doing the test once again now.  All this is bothersome because I thought the issue was solved some time ago and I am not able to understand why it is back for me.  I have tried to make sure that the version of Mirage being used is what I think it is, namely an installation I did on 14 January.  But I worry a bit that conda is doing something that causes me to not have the version I think I should have in the run.

KevinVolkSTScI, 2021-01-25:
It looks like the conda version is 1.3.3 for both tests and the WFSS/F090W test does not have the direct seed images covering the entire POM area.  Is there a chance that this is due to somehow having an out-of-date set-up for the various files that Mirage needs?  I am pointing to /ifs/jwst/wit/mirage_data/ as usual for the data files.

bhilbert4, 2021-03-30:
Hi Kevin. Did you resolve this problem? I agree that I think the issue was the version of the code being used. 1.3.3 doesn't use transmission files at all, so the seed images it produces will be a different size compare to those from the newer code. 

To check what version of Mirage you are running:

import mirage
mirage.__version__

I plan on making a new release of Mirage soon. I'm waiting on some work on the disperser. It seems to be taking a lot of memory and running slower than it used to. Nor is looking into it. Once that is corrected I'll release Mirage version 2.0.

bhilbert4, 2021-03-30:
You should always deactivate the current environment before activating a new one. If you don't, then both environments can be active at the same time, and code from both can be accessed.

KevinVolkSTScI, 2021-05-28:
I ran another test today using a recent version of Mirage and again the code refuses to simulate the POM area.  In this case I ran two .yaml files through Mirage one with mode = imaging and one with mode = pom and the seed images out were in both cases 2048x0248 pixels.  This is getting to be a problem for simulating NIRISS commissioning data.

bhilbert4, 2021-05-28:
Now that we've separated the application of the flat field from that of the POM transmission image, it looks like this mode has accidentally been lumped in with imaging mode, where everything is cropped to 2048x2048. It's easy enough to turn off cropping, but the harder part that I see now is that we don't have POM transmission images for filters in the filter wheel for NIRISS. It looks like the filter dependence is rather small. Should we just use the image for F200W for all of the filters in the filter wheel? Or is it worth generating new images for those filters? See the existing POM transmission images here (*pomtransmission*fits): https://github.com/npirzkal/GRISM_NIRISS/tree/master/V3

KevinVolkSTScI, 2021-05-28:
I do not understand these POM transmission images.  For one thing, the area is larger than the POM and signal from the outer areas will not affect the images whether in imaging or WFSS mode.  It cannot affect it, since the light from those pixels never reaches the NIRISS internal optics.  Second, the features in the POM image are already in the flat fields one gets from CRDS (e.g. the POM feature at about [1570, 1045] and the spots).  But I assume Chris looked at the images and if he is OK with them then things are presumably being handled properly.
We do need to generate these files for the other filters at longer wavelengths if they are used, since the features continue to decline in strength at longer wavelengths.

KevinVolkSTScI, 2021-05-28:
I suppose that these POM images are used to remove the POM features from the flat fields in the WFSS case.  They should not be applied in the imaging case.  For imaging this image should be just 1.0 in all pixels.  The same is true for the POM case, the POM image should just be all 1.0 if it is applied.  It is only in the dispersed images that these features need to be removed from the flat fields.

bhilbert4, 2021-05-28:
Ok, that should be an easy fix then. I can generate an array of 1s for this case. That is what is done in the case of imaging mode simulations, but in that case the image is 2048x2048. So for POM mode I can just create a larger array. Do you want me to go back to the size of the array that was used previously: 2322x2322?

KevinVolkSTScI, 2021-05-28:
Yes, the size should be 2322x2322 and the central 2048x2048 pixels corresponds to the imaging area.  (The real situation is complex because the image area is both rotated with respect to the POM edges and offset from the centre of the POM area, but that is too complex to deal with here I would think.)

bhilbert4, 2021-05-28:
That's why we have those images for WFSS simulations. Those, I believe, do capture the rotation.

KevinVolkSTScI, 2021-05-28:
This poses a question.  If somehow the WFSS POM images allow for the rotation and offset presumably they have to have zeros for the pixels mapping beyond the POM edges.  That would suggest that the same should be done for the POM mode and imaging cases otherwise things are inconsistent between where sources appear for the WFSS dispersed case and the POM/imaging direct image case.
Can you make an image for the POM mode case where the "central" pixels are all 1 and anything beyond the edges are 0?

bhilbert4, 2021-05-28:
Wouldn't that just be the same as the WFSS POM images? Or at least, the central 2322x2322 pixels of the WFSS POM images? If you use the link above, or look in the GRISM_NIRISS/current/ directory in your collection of Mirage reference files, you'll be able to see the WFSS POM images.

KevinVolkSTScI, 2021-05-28:
Yes it should be the same, just the features within the "active" area would not apply for POM or imaging mode.  Whether it is the central area or not depends on how the larger POM transmission images were created.

KevinVolkSTScI, 2021-05-28:
What I mean is, I do not know off hand whether the POM area is centred in the larger POM transmission image or not.  Or whether the 2048x2048 pixel detector area is centered within the larger images.  I could find out, but it is not obvious just looking casually at the POM transmission images.

PR #681 (2021-05-28): Keep POM mode simulation seed images as 2322x2322 pix
This PR fixes an oversight from #523  where POM transmission images were introduced. For "pom" mode simulations in Mirage, the desired output is a seed image that is larger than the detector, so that sources just off the edge of the detector can be seen. When we introduced the POM transmission images, the pom mode simulations were lumped in with imaging mode simulations, which are cut down to the detector size of 2048x2048 pixels.

This PR reverses that, and returns NIRISS pom mode simulations to 2322x2322 pixels. Note that the pom mode see [... cut]
closed 2021-05-28 by bhilbert4