spacetelescope/mirage #635: Source Catalogs have no magnitude specifically called nircam_f182m_magnitude

eas342, 2021-02-18:
I was running the MIRAGE notebook today to make the accompanying SW imaging data that comes with the grism spectroscopy. I came across this error running the notebook here:
`https://github.com/spacetelescope/mirage/blob/master/examples/NIRCam_TSO_examples.ipynb`

It happens at this step
```python
img_tso = ImgSim()
img_tso.paramfile = img_tso_sw_yaml
img_tso.create()
```

> ValueError: WARNING: Catalog tso_imaging_source.cat has no magnitude column for NIRCAM specifically called nircam_f182m_magnitude, nor a generic 'magnitude' column. Unable to proceed.

I can make it go away by explicitly creating a column in `ptsrcs.cat` and `tso_imaging_source.cat` that has the header `nircam_f182m_magnitude` with a copy of the magnitude from `nircam_f182m_clear_magnitude`

bhilbert4, 2021-02-26:
Have you installed Mirage relatively recently? I put in some logic to deal with cases like this, probably a few months ago. If e.g. nircam_f182m_clear_magnitude is not there, then it should fall back to look for nircam_f182m_magnitude. https://github.com/spacetelescope/mirage/blob/master/mirage/seed_image/catalog_seed_image.py#L3597

bhilbert4, 2021-03-24:
@eas342 is this still an issue for you?

eas342, 2021-03-24:
Yes, I still got this yesterday with the TSO imaging section of the example TSO notebook. I tried installing mirage on a high performance cluster yesterday and will see if the problem happens there too once I figure out the right slurm script.

eas342, 2021-03-26:
I just ran this on the high performance cluster and have the same problem. My `ptsrcs.cat` looks like:

```
# position_RA_Dec
# vegamag
# 
# 
index x_or_RA y_or_Dec nircam_f182m_clear_magnitude nircam_f444w_clear_magnitude nircam_f322w2_clear_magnitude nircam_f210m_clear_magnitude nircam_f444w_f470n_magnitude
1 66.378244 -30.598312 13.75 12.5 12.75 13.39 12.4
```

<details>
  File "/xdisk/eas342/eas342/mirage_examples/mirage/logging/logging_functions.py", line 111, in wrapped
    func(*args, **kwargs)
  File "/xdisk/eas342/eas342/mirage_examples/mirage/seed_image/catalog_seed_image.py", line 364, in make_seed
    self.seedimage, self.seed_segmap = self.create_sidereal_image()
  File "/xdisk/eas342/eas342/mirage_examples/mirage/seed_image/catalog_seed_image.py", line 2252, in create_sidereal_image
    segment_offset=offset_vector)
  File "/xdisk/eas342/eas342/mirage_examples/mirage/seed_image/catalog_seed_image.py", line 2642, in get_point_source_list
    mag_column = self.select_magnitude_column(lines, filename)
  File "/xdisk/eas342/eas342/mirage_examples/mirage/seed_image/catalog_seed_image.py", line 3635, in select_magnitude_column
    specific_mag_col)))
ValueError: WARNING: Catalog ptsrcs.cat has no magnitude column for NIRCAM specifically called nircam_f182m_magnitude, nor a generic 'magnitude' column. Unable to proceed.

</details>

bhilbert4, 2021-03-26:
I found the problem with this. The TSIMAGE observation uses F182M/WLP8, but the catalog creation process adds a magnitude column for F182M/CLEAR. So Mirage looks at the catalog, sees that none of the columns, including F182M/CLEAR match what it's looking for, and so it falls back to look for nircam_f182m_magnitude. When it doesn't find that, it fails. So the answer is to change the calls to 'add_magnitude_column' so that they create nircam_f182m_wlp8_magnitude columns. Note that at the wavelength of F182M, the throughput for WLP8 is very close to, but not quite, 1.0, so depending on how accurate you want to be, you might have to adjust the magnitude values that are put in the catalogs. I have this fix in PR #650, along with a small change to the way Mirage looks for catalog column names. (No more falling back to e.g. nircam_f182m_magnitude when weak lenses are used). I'll merge that today, and then the notebook in the repo will be updated. It's a pretty trivial change though, if you want to reproduce it in your notebook.

PR #650 (2021-03-26): Look for correct WLP8 catalog colname in TSO notebook
Resolves #635 

This PR makes some small changes to the case where a weak lens is used in a source catalog.

- In cases where a weak lens is used, if the corresponding catalog column name nircam\_\<filter>\_wlp8\_magnitude or nircam\_\<filter>\_wlm8\_magnitude is not present, then an exception is raised. Previously Mirage would fall back to look for nircam\_\<filter>\_magnitude, however the +/-8 weak lenses have a significant effect on throughput and should not be ignored. This was also leading to confusing errors in the case where nircam\_ [... cut]
closed 2021-03-26 by bhilbert4