spacetelescope/mirage #640: Connection reset by peer during Gaia lookup

eas342, 2021-03-12:
I got this error when trying the automatic catalog creation.

``` python
import os
import numpy as np
from mirage.catalogs import catalog_generator
from mirage.catalogs import create_catalog

object_ra = (10. + 19./60. + 37.96/3600.) * 15.
object_dec = -1.0 * (9. + 48./60. + 23.20/3600.)

filter_list = ['F444W', 'F322W2','F182M','F210M','F470N']
box_width=1 ## arcsec
#                                                        besancon_catalog_file=besancon_result,

cat, mag_column_names = create_catalog.get_all_catalogs(object_ra, object_dec, box_width,
                                                        instrument='NIRCAM', filters=filter_list,
                                                        ra_column_name='RAJ2000', dec_column_name='DECJ2000',
                                                        starting_index=1)
```

I think this might be similar to the issue you had a little while ago (https://github.com/astropy/astroquery/issues/1524), but it happens with any `box_size` to me. Maybe it's just a server problem?

<details>
  <summary> Expand for full traceback </summary>
```python
2021-03-11 21:25:46,455 - mirage.catalogs.create_catalog.query_GAIA_ptsrc_catalog - INFO - Searching the GAIA DR2 catalog
---------------------------------------------------------------------------
ConnectionResetError                      Traceback (most recent call last)
<ipython-input-3-4e2185f7a390> in <module>
      9                                                         instrument='NIRCAM', filters=filter_list,
     10                                                         ra_column_name='RAJ2000', dec_column_name='DECJ2000',
---> 11                                                         starting_index=1)

~/miniconda3/envs/mirage/lib/python3.6/site-packages/mirage/catalogs/create_catalog.py in get_all_catalogs(ra, dec, box_width, besancon_catalog_file, instrument, filters, ra_column_name, dec_column_name, starting_index)
    595 
    596     gaia_cat, gaia_mag_cols, gaia_2mass, gaia_2mass_crossref, gaia_wise, \
--> 597         gaia_wise_crossref = query_GAIA_ptsrc_catalog(outra, outdec, box_width)
    598     twomass_cat, twomass_cols = query_2MASS_ptsrc_catalog(outra, outdec, box_width)
    599     wise_cat, wise_cols = query_WISE_ptsrc_catalog(outra, outdec, box_width)

~/miniconda3/envs/mirage/lib/python3.6/site-packages/mirage/catalogs/create_catalog.py in query_GAIA_ptsrc_catalog(ra, dec, box_width)
   1862     logger.info('Searching the GAIA DR2 catalog')
   1863     for key in data.keys():
-> 1864         job = Gaia.launch_job_async(data[key]['query'], dump_to_file=False)
   1865         table = job.get_results()
   1866         outvalues[key] = table

~/miniconda3/envs/mirage/lib/python3.6/site-packages/astroquery/utils/tap/core.py in launch_job_async(self, query, name, output_file, output_format, verbose, dump_to_file, background, upload_resource, upload_table_name, autorun)
    400                                         verbose,
    401                                         name,
--> 402                                         autorun)
    403         isError = self.__connHandler.check_launch_response_status(response,
    404                                                                   verbose,

~/miniconda3/envs/mirage/lib/python3.6/site-packages/astroquery/utils/tap/core.py in __launchJob(self, query, outputFormat, context, verbose, name, autorun)
    609         response = self.__connHandler.execute_tappost(subcontext=context,
    610                                                       data=data,
--> 611                                                       verbose=verbose)
    612         if verbose:
    613             print(response.status, response.reason)

~/miniconda3/envs/mirage/lib/python3.6/site-packages/astroquery/utils/tap/conn/tapconn.py in execute_tappost(self, subcontext, data, content_type, verbose)
    271         """
    272         context = self.__get_tap_context(subcontext)
--> 273         return self.__execute_post(context, data, content_type, verbose)
    274 
    275     def execute_datapost(self, data,

~/miniconda3/envs/mirage/lib/python3.6/site-packages/astroquery/utils/tap/conn/tapconn.py in __execute_post(self, context, data, content_type, verbose)
    402         self.__postHeaders["Content-type"] = content_type
    403         conn.request("POST", context, data, self.__postHeaders)
--> 404         response = conn.getresponse()
    405         self.__currentReason = response.reason
    406         self.__currentStatus = response.status

~/miniconda3/envs/mirage/lib/python3.6/http/client.py in getresponse(self)
   1352         try:
   1353             try:
-> 1354                 response.begin()
   1355             except ConnectionError:
   1356                 self.close()

~/miniconda3/envs/mirage/lib/python3.6/http/client.py in begin(self)
    305         # read until we get a non-100 response
    306         while True:
--> 307             version, status, reason = self._read_status()
    308             if status != CONTINUE:
    309                 break

~/miniconda3/envs/mirage/lib/python3.6/http/client.py in _read_status(self)
    266 
    267     def _read_status(self):
--> 268         line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
    269         if len(line) > _MAXLINE:
    270             raise LineTooLong("status line")

~/miniconda3/envs/mirage/lib/python3.6/socket.py in readinto(self, b)
    584         while True:
    585             try:
--> 586                 return self._sock.recv_into(b)
    587             except timeout:
    588                 self._timeout_occurred = True

~/miniconda3/envs/mirage/lib/python3.6/ssl.py in recv_into(self, buffer, nbytes, flags)
   1010                   "non-zero flags not allowed in calls to recv_into() on %s" %
   1011                   self.__class__)
-> 1012             return self.read(nbytes, buffer)
   1013         else:
   1014             return socket.recv_into(self, buffer, nbytes, flags)

~/miniconda3/envs/mirage/lib/python3.6/ssl.py in read(self, len, buffer)
    872             raise ValueError("Read on closed or unwrapped SSL socket.")
    873         try:
--> 874             return self._sslobj.read(len, buffer)
    875         except SSLError as x:
    876             if x.args[0] == SSL_ERROR_EOF and self.suppress_ragged_eofs:

~/miniconda3/envs/mirage/lib/python3.6/ssl.py in read(self, len, buffer)
    629         """
    630         if buffer is not None:
--> 631             v = self._sslobj.read(len, buffer)
    632         else:
    633             v = self._sslobj.read(len)

ConnectionResetError: [Errno 104] Connection reset by peer

```
</details>

bhilbert4, 2021-03-12:
Interesting. I'm not sure. Are you pointing at the galactic center? :)

eas342, 2021-03-12:
Nope, not near the galactic center or particularly dense. The problem went away this morning so I'm guessing it was a server issue.

closed 2021-03-12 by eas342