spacetelescope/mirage #641: Transmission reffiles are null

eas342, 2021-03-12:
I ran the latest development version of `mirage` today for TSO simulations and got an AttributeError that I traced back to the TSO YAML files.

The input YAML file has the following Reffiles dictionary:
``` python
Reffiles:
  astrometric: /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_distortion_0090.asdf
  badpixmask: /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_mask_0038.fits
  crosstalk: /home/schlawin/other_code/mirage/mirage/config/xtalk20150303g0.errorcut.txt
  dark: None
  filter_throughput: /home/schlawin/other_code/mirage/mirage/config/placeholder.txt
  filter_wheel_positions: /home/schlawin/other_code/mirage/mirage/config/nircam_filter_and_pupil_wheel_positions.txt
  filtpupilcombo: /home/schlawin/other_code/mirage/mirage/config/nircam_filter_pupil_pairings.list
  flux_cal: /home/schlawin/other_code/mirage/mirage/config/NIRCam_zeropoints.list
  gain: /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_gain_0056.fits
  illumflat: None
  invertIPC: false
  ipc: /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/Kernel_to_add_IPC_effects_from_jwst_nircam_ipc_0028.fits
  linearity: /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_linearity_0052.fits
  linearized_darkfile: /fenrirdata1/program_data/mirage_data/nircam/darks/linearized/A5/NRCNRCALONG-DARK-53510917431_1_485_SE_2015-12-17T10h15m23_uncal_seg001_linear_dark_prep_object.fits
  occult: None
  photom: /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_photom_0071.fits
  pixelAreaMap: /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_area_0015.fits
  pixelflat: /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_flat_0309.fits
  readpattdefs: /home/schlawin/other_code/mirage/mirage/config/nircam_read_pattern_definitions.list
  saturation: /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_saturation_0063.fits
  subarray_defs: /home/schlawin/other_code/mirage/mirage/config/NIRCam_subarray_definitions.list
  superbias: /fenrirdata1/es_tso/crds_cache/references/jwst/nircam/jwst_nircam_superbias_0022.fits
  transmission: null
```
Then, line 510 in "utils.py" gets an attribute error because `if params[key1][key2].lower() not in ['none', 'config', 'crds']:` NoneType cannot be lower(). I was able to fix this by making a switch to set them to 'crds'.

Should it be set to 'crds' or something instead of `null`? Let me know if you need a working example. Sorry, this was kind of hard to make minimal working examples for because the problem happens way down in the TSO notebook.

bhilbert4, 2021-03-12:
Hmm, yes, that should be 'crds'. Or, if you specify 'crds', Mirage should translate that to a full file path, like it does for the other files. I'll have to look into why that's different for the transmission file. I wonder if there's some difference between my environment and yours in terms of how Mirage is looking at the CRDS cache.

eas342, 2021-03-12:
Ok, I made a minimal-ish example below. I could've sworn "transmission" was "null" yesterday but now the attribute error seems to arise somewhere else.

[pointing_and_xml.zip](https://github.com/spacetelescope/mirage/files/6132271/pointing_and_xml.zip)

```python
from mirage.yaml import yaml_generator
from mirage.catalogs.catalog_generator import GrismTSOCatalog
from mirage.grism_tso_simulator import GrismTSO
import yaml
inputDir = '/fenrirdata1/es_tso/sim_data/mirage_tso_notebook/input_010_shorter_still_ers/'
xml_file = inputDir + 'ERS_WASP43_even_shorter.xml'
pointing_file = inputDir + 'ERS_WASP43_even_shorter.pointing'
output_data_dir = '/fenrirdata1/es_tso/sim_data/mirage_tso_notebook/files_to_test_mirage/output/'
output_yaml_dir = '/fenrirdata1/es_tso/sim_data/mirage_tso_notebook/files_to_test_mirage/output/'
background='medium'
pav3 = 0.
dates = '2022-04-25'

yam = yaml_generator.SimInput(xml_file, pointing_file,  verbose=False,
                              output_dir=output_yaml_dir, simdata_output_dir=output_data_dir,
                              background=background,
                              roll_angle=pav3,
                              dates=dates, datatype='linear, raw', dateobs_for_background=True,
                              reffile_defaults='crds')
yam.create_inputs()
gr_tso_yaml_file = 'output/jw00042001001_01101_00002_nrca5.yaml'
gr_f322w2 = GrismTSO(gr_tso_yaml_file,# SED_file=sed_file, SED_normalizing_catalog_column=None,
                    final_SED_file=None, save_dispersed_seed=True, source_stamps_file=None,
                    extrapolate_SED=True, override_dark=None, disp_seed_filename=None,
                    orders=["+1", "+2"])
gr_f322w2.create()

```

<details>
<summary> Click for traceback </summary>
2021-03-12 11:41:21,024 - root - CRITICAL - Traceback (most recent call last):
  File "/home/schlawin/other_code/mirage/mirage/logging/logging_functions.py", line 111, in wrapped
    func(*args, **kwargs)
  File "/home/schlawin/other_code/mirage/mirage/seed_image/catalog_seed_image.py", line 175, in make_seed
    self.params = utils.full_paths(self.params, self.modpath, self.crds_dict, offline=self.offline)
  File "/home/schlawin/other_code/mirage/mirage/utils/utils.py", line 510, in full_paths
    if params[key1][key2].lower() not in ['none', 'config', 'crds']:
AttributeError: 'NoneType' object has no attribute 'lower'

2021-03-12 11:41:21,026 - root - CRITICAL - CRASHED
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
<ipython-input-21-00439f32edd1> in <module>
----> 1 gr_f322w2.create()

~/other_code/mirage/mirage/grism_tso_simulator.py in create(self)
    261         background_direct = catalog_seed_image.Catalog_seed()
    262         background_direct.paramfile = self.background_paramfile
--> 263         background_direct.make_seed()
    264         background_segmentation_map = background_direct.seed_segmap
    265 

~/other_code/mirage/mirage/logging/logging_functions.py in wrapped(*args, **kwargs)
    109         try:
    110             # Run the function
--> 111             func(*args, **kwargs)
    112             #logging.info('Completed Successfully')
    113 

~/other_code/mirage/mirage/seed_image/catalog_seed_image.py in make_seed(self)
    173 
    174         # Expand param entries to full paths where appropriate
--> 175         self.params = utils.full_paths(self.params, self.modpath, self.crds_dict, offline=self.offline)
    176         self.filecheck()
    177         self.basename = os.path.join(self.params['Output']['directory'],

~/other_code/mirage/mirage/utils/utils.py in full_paths(params, module_path, crds_dictionary, offline)
    508                     params[key1][key2] = 'crds'
    509 
--> 510             if params[key1][key2].lower() not in ['none', 'config', 'crds']:
    511                 params[key1][key2] = os.path.abspath(os.path.expandvars(params[key1][key2]))
    512             elif params[key1][key2].lower() == 'config':

AttributeError: 'NoneType' object has no attribute 'lower'
</details>

bhilbert4, 2021-03-17:
Can you send me (here or via email) the batman parameters or lightcurves that you are using? I'm putting together an example run, but so far my choices of parameters have been bad.

eas342, 2021-03-17:
Here were the parameters I used
``` python
# Esposito et al. 2017 parameters on NASA Exoplanet Archive
params = batman.TransitParams()       # object to store transit parameters 
params.t0 = 4320.                        # time of inferior conjunction (This one is just offset from start)
params.per = 0.813473978 * 24. * 3600.    # orbital period, Esposito
params.rp = 0.1588                       # planet radius (in units of stellar radii) , Esposito
params.a = 4.97                         # semi-major axis (in units of stellar radii), Esposito
params.inc = 82.109                     # orbital inclination (in degrees) , Esposito
params.ecc = 0.                       # eccentricity 
params.w = 90.                        # longitude of periastron (in degrees) 
params.limb_dark = "nonlinear"        # limb darkening model 
params.u = [0.3, 0.1, 0.1, -0.1]      # limb darkening coefficients [u1, u2, u3, u4] 
```

but I don't think the light curves matter for this particular error. The "minimal-ish" example above doesn't even use a lightcurve I don't think.

bhilbert4, 2021-03-17:
I was wondering about that. My test seems to have gotten beyond where you saw the error, because it was failing on the bad lightcurves that I had created.

eas342, 2021-03-17:
FYI, I have been able to get `mirage` to run without the "AttributeError" by modifying `mirage.utils.utils` like so:
https://github.com/spacetelescope/mirage/compare/master...eas342:attributeErrorInParams

bhilbert4, 2021-03-17:
Can you send me the yaml file that this is happening with? Somehow one of the entries is being read in as None rather than 'None'. I'm not sure why that is though.

eas342, 2021-03-17:
It looks like the `_tso_grism_sources.yaml` and `background_sources.yaml` have `null` for `transmission`.
[jw00042001001_01101_00002_nrca5_background_sources.yaml.txt](https://github.com/spacetelescope/mirage/files/6159881/jw00042001001_01101_00002_nrca5_background_sources.yaml.txt)
[jw00042001001_01101_00002_nrca5_tso_grism_sources.yaml.txt](https://github.com/spacetelescope/mirage/files/6159883/jw00042001001_01101_00002_nrca5_tso_grism_sources.yaml.txt)

bhilbert4, 2021-03-17:
Strange. And those files were created by the call to the yaml_generator above? It looks like there is a disconnect between the files created by the yamnk_generator and the file fed into the call to GrismTSO, but I'm guessing that was part of the minimum working example creation?

bhilbert4, 2021-03-17:
Oh wait. I see. These are the yaml files created internally from the input yaml file. Could you send me output/jw00042001001_01101_00002_nrca5.yaml?

eas342, 2021-03-17:
That one seems ok. No `null`
[jw00042001001_01101_00002_nrca5.yaml.txt](https://github.com/spacetelescope/mirage/files/6159965/jw00042001001_01101_00002_nrca5.yaml.txt)

bhilbert4, 2021-03-18:
Do you have the transmission reference files? Nor and I had to reorganize things a bit a while ago in order to support different versions of the NIRCam and NIRISS files. They should be called *POM*fits and be sitting in $MIRAGE_DATA/nircam/GRISM_NIRCAM/current/

https://mirage-data-simulator.readthedocs.io/en/latest/reference_files.html?highlight=GRISM_NIRCAM#download-grism-related-reference-data

If you don't have those, it would be worth grabbing them and also re-installing NIRCam_Gsim, grismconf, and Mirage. The updates to all 3 packages went hand-in-hand. If those files aren't there, that's probably why this null value is showing up. I'll add a check in the code for this and raise an exception if the transmission files are not present.

bhilbert4, 2021-03-18:
If you look in the log file from your Mirage run, there should be a line in there that says: "From grism library, found {} as the POM transmission file". This will show whether a real file was found or not.

eas342, 2021-03-18:
Aha, thanks for that tip @bhilbert4! I updated my GRISM_NIRCAM reference data and now the the `tso_grism_sources.yaml` is whole again (ie. the transmission points to a file instead of `null`).

closed 2021-03-18 by eas342