spacetelescope/mirage #648: Flux normalization is off

eas342, 2021-03-24:
I ran the NIRCam TSO examples notebook here and the flux normalization appears way off:
https://github.com/spacetelescope/mirage/blob/master/examples/NIRCam_TSO_examples.ipynb

The image is way saturated on the first group. Here are the versions of stuff I have:
* mirage '1.3.4.dev467+g7e275b7'
* NIRCam-Gsim 1.5 and latest development version both had the problem
* GRISM_NIRCAM/V2 data

[image]

Any other package versions or data files I should check?

bhilbert4, 2021-03-24:
@NorPirzkal

bhilbert4, 2021-03-24:
@eas342 how do the fluxes look in the undispersed seed image?  And are you providing an SED for the target with values and units that are correct? I'm assuming that this is a case that used to work and that inputs haven't changed, but I just wanted to be sure.

eas342, 2021-03-24:
The `tso_grism_sources_F444W_GRISMR_final_seed_image` between an old mirage run and the new one look about the same (other than different image dimensions). The `_dispersed_seed_images.fits` is ~40,000 times larger in the new mirage run as compared to the old one if that's a clue. I applied normalization with the values from the notebook linked at the top Johnson K=9.06 and then again with object_f444w_mag = 9.0.

NorPirzkal, 2021-03-24:
Hi Everett,

What are the range of values in your input spectrum and what units are you using?
What do you get if you do not use your own spectrum and instead let Mirage use the photometry to determine a low res SED?

N

eas342, 2021-03-24:
Thanks for looking into this @NorPirzkal ! Some of what you need might be hidden from foolish users like me under the hood of `mirage`. In the beginning of the example mirage notebook, it sets wavelength units to microns and flux units to `flam`, which ranges from 0 to 1e-13. Can you help point me to which file or input would be useful to check the counts in for NIRCam_Gsim? There is a `source_sed_file_from_ptsrcs_and_test_grism_tso_sed_file_wasp79.hdf5`

eas342, 2021-03-26:
I installed everything on a compute cluster to test this and the problem recurs. One thing I was wondering if there could be an issue with my reference data? I did copy all of my MIRAGE_DATA, GRISM_NIRCAM/V2 and CDBS data from the other server.

eas342, 2021-03-29:
I also tried rolling back NIRCAM_GSIM to version 1.4b1 along with using GRISM_NIRCAM/V1 and the first group is still way saturated in the simulation. So, the solution probably needs a more intelligent approach of testing the dispersion step than my wild guesses about package versions and grism data.

eas342, 2021-04-02:
Any suggestions? Fortunately a student kept older versions of mirage from early 2020 before upgrading so we will use that for now.

bhilbert4, 2021-04-06:
I'm trying a test case now. Using python 3.8 I got a slew of multiprocessing errors during dispersion. I backed down to python 3.6 and dispersion is working, but is slow. I'll report back when it finishes up.

eas342, 2021-04-06:
Yay! The normalization problems looks to be fixed by https://github.com/npirzkal/NIRCAM_Gsim/commit/39ea72cbeea0e378c120a7b3c82a7fe9bd349565

closed 2021-04-06 by eas342