spacetelescope/mirage #666: Excesive log entries when a filter has no GAP value.

KevinVolkSTScI, 2021-04-21:
I am running a simulation of the astrometric field with the current version of Mirage installed today.  The first image in the simulation is for the F277W filter for which there is no ghost axis point defined.  This has resulted in 4.6 million lines in the log file, almost all of which are noting that the F277W filter has not GAP value and ghosts are not going to be used.  (By the way, I had not intended to have the ghost option on anyway.  I had assumed it would be off by default rather than on, because one really needs to have ghost images for each point source to do it properly, and I have a large number of sources to deal with.)

I would think it more reasonable to report this error once in the log file when a point source is processed for the source catalogue, and then not to repeat it N times more for however many stars are in the list.

Also, when I read the documentation page for the ghosts in NIRISS it sounded like all filters can use this option.  I see that the ghosts file contains 5 filters (F090W, F115W, F140M, F150W, and F200W) but lacks the other seven filters.  This probably should be made clear to users.

bhilbert4, 2021-04-21:
It sounds like your version of the code is a bit behind. I fixed this logging error in #636 which was merged on Feb 25.

As for ghosts being on by default, I figured if the goal of Mirage is to create data that is as realistic as possible, it makes sense to add ghosts when possible. Are you looking to avoid ghosts for your field in the interest of computation time?

I'll update the documentation to explicitly mention the filters for which ghosts are added.

KevinVolkSTScI, 2021-04-21:
I installed the code today following the standard instructions so I do not understand why the version is out of date.

Yes with the astrometric field I did not want to add ghosts because of the number of sources involved.

bhilbert4, 2021-04-21:
Check the version number with `mirage.__version__`.  Depending on exactly when you installed today, you should have 2.0.1 or 2.0.2. If you do, let me know. Maybe my fix for this issue was incomplete.

KevinVolkSTScI, 2021-04-21:
The version is 2.0.2.

PR #667 (2021-04-23): Log the absence of a filter from the ghost gap file only once
Resolves #666 

This PR adjusts the logging in catalog_seed_generator.py such that in cases where ghosts are requested, if the filter used does not support ghosts, only one log entry is made saying so.
closed 2021-04-23 by bhilbert4