spacetelescope/mirage #672: WFSS simulations failing in version 2.0.2

bhilbert4, 2021-04-29:
@swara13 @jotaylor @mtakahiro report that WFSS simulations are crashing in Mirage 2.0.2:

```
 44%|████▍     | 8/18 [01:25<01:21,  8.12s/it]/Users/swara/miniconda3/envs/mirage_april2021/lib/python3.7/site-packages/scipy/interpolate/interpolate.py:623: RuntimeWarning: divide by zero encountered in true_divide
  slope = (y_hi - y_lo) / (x_hi - x_lo)[:, None]
/Users/swara/miniconda3/envs/mirage_april2021/lib/python3.7/site-packages/scipy/interpolate/interpolate.py:623: RuntimeWarning: divide by zero encountered in true_divide
  slope = (y_hi - y_lo) / (x_hi - x_lo)[:, None]
/Users/swara/miniconda3/envs/mirage_april2021/lib/python3.7/site-packages/scipy/interpolate/interpolate.py:623: RuntimeWarning: divide by zero encountered in true_divide
  slope = (y_hi - y_lo) / (x_hi - x_lo)[:, None]
/Users/swara/miniconda3/envs/mirage_april2021/lib/python3.7/site-packages/scipy/interpolate/interpolate.py:623: RuntimeWarning: divide by zero encountered in true_divide
  slope = (y_hi - y_lo) / (x_hi - x_lo)[:, None]
```

```Dispersing order  +2
100%|██████████| 18/18 [02:18<00:00,  7.70s/it]
  0%|          | 0/18 [00:00<?, ?it/s]
Dispersing order  +3
100%|██████████| 18/18 [02:25<00:00,  8.06s/it]
  0%|          | 0/18 [00:00<?, ?it/s]
Dispersing order  -1
100%|██████████| 18/18 [02:07<00:00,  7.10s/it]
Computing dispersed background for order  +1
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
<ipython-input-36-6807b5dc9df4> in <module>
      4                            SED_file=None, SED_normalizing_catalog_column='niriss_f200w_magnitude',
      5                            SED_dict=my_sed, create_continuum_seds=True)
----> 6 m.create()
~/miniconda3/envs/mirage_april2021/lib/python3.7/site-packages/mirage/wfss_simulator.py in create(self)
    219                 # Only include the background in one of the object type seed images
    220                 if not background_done:
--> 221                     background_image = dispersed_objtype_seed.disperse_background_1D([back_wave, back_sig])
    222                     scaling_factor = np.max(background_image)
    223                     dispersed_objtype_seed.finalize(BackLevel=scaling_factor, tofits=self.background_image_filename)
~/miniconda3/envs/mirage_april2021/lib/python3.7/site-packages/NIRCAM_Gsim/grism_seed_disperser.py in disperse_background_1D(self, background)
    154         for order in self.orders:
    155             print("Computing dispersed background for order ",order)
--> 156             bck += self.this_one[order].disperse_background_1D(background)
    157 
    158 #        fits.writeto("WFSS_background.fits",bck,overwrite=True)
~/miniconda3/envs/mirage_april2021/lib/python3.7/site-packages/NIRCAM_Gsim/observations/observations.py in disperse_background_1D(self, background)
    353             #print("disperse_background_1D: x-direction")
    354             direction = "x"
--> 355             xs = np.arange(self.C.XRANGE[self.order][0]+0,self.C.XRANGE[self.order][1]+naxis[0])
    356             ys = np.zeros(np.shape(xs))+ypos
    357         else:
TypeError: 'NoneType' object is not subscriptable
```

bhilbert4, 2021-05-23:
I get the error above when running with python 3.6. With python 3.8 I get a crazy multiprocessing error, even though I'm running from a script rather than a notebook. @NorPirzkal any ideas, at least on the error above?

bhilbert4, 2021-05-23:
@NorPirzkal It looks like the problem is that the latest version of the NIRISS conf files are missing the XRANGE and YRANGE fields.

npirzkal, 2021-05-24:
@bhilbert4 yes, I see how I use the XRANGE value when dispersing the 1D background to get the initial estimate. I have added the keywords in the NIRISS config files for now.

bhilbert4, 2021-05-24:
@swara13 @jotaylor @mtakahiro See if the changes Nor made have solved the issue. I updated the reference files in the central store, so you should be able to re-run your code without having to update anything.

mtakahiro, 2021-05-24:
@bhilbert4 @NorPirzkal Thanks! I just ran the same notebook and could run through without having any issues. Would be great if either @jotaylor or @swara13 can confirm this when they have time.