spacetelescope/mirage #677: Ramp generator stitching file segments incorrectly

hover2pi, 2021-05-05:
With long observations `obs_generator` seems to be saving file segments incorrectly. For example, a 6 group, 320 integration simulation of NIRISS SOSS data generates two segments of data of 213 and 107 integrations. Since this data contains a planetary transit that happens to fall right in the middle of the file split, the reduced data shows this problem.

[image]

A quick look at the saved seed image shows this does not appear in the raw data.

[image]

So perhaps the `seg002` file, which has no visible transit, isn't being saved properly. One theory is that `obs_genetrator` is slicing the seed image from the beginning of the observation for all segments.

bhilbert4, 2021-05-07:
Hmm. My first thought is that this is SOSS specific somehow. NIRCam TSO observations that are split into segments are stitched back together correctly from what I have seen. I wonder if this is related to that if/else clause in #671, in obs_generator.py around line 200. Maybe I can try to dig up a TSO observation and we can compare the intermediate files (seed image files, darks, etc)?

hover2pi, 2021-05-14:
Yeah, I'll put the error catch back in there and try to fix it another way to see if the issue is resolved.

eas342, 2021-09-17:
@hover2pi , did you end up resolving this?

hover2pi, 2022-04-08:
Yes!

closed 2022-04-08 by hover2pi