spacetelescope/mirage #693: Transmission set to None in yaml file for NIRCam imaging

micaelabagley, 2021-06-17:
I'm getting an AttributeError related to the transmission reffiles, using Mirage v2.0.2 and yamls created using `yaml_generator` with `reffile_defaults='crds_full_name'` and an APT pointing and xml file. This is similar to issue #641, but I am only using NIRCam Imaging mode (no WFSS observations). 
The yaml files have full path names for the reffiles, but for transmission they have `transmission: None`. It runs fine when I replace `None` with `crds`.

In the log file, it does indicate that it can't find a POM transmission file:
```
2021-06-17 09:58:33,959 - mirage.reference_files.utils.get_transmission_file - INFO - POM Transmission filename: None
2021-06-17 09:58:33,960 - mirage.utils.utils.full_paths - INFO - From grism library, found None as the POM transmission file.
...
2021-06-17 09:59:21,342 - mirage.seed_image.catalog_seed_image - INFO - No transmission file given. Assuming full transmission for all pixels, not including flat field effects. (no e.g. occulters blocking any pixels).
2021-06-17 09:59:21,342 - stpipe - INFO - No transmission file given. Assuming full transmission for all pixels, not including flat field effects. (no e.g. occulters blocking any pixels).
```

Do I need the full suite of Grism reffiles to run Mirage for NIRCam Imaging mode?

bhilbert4, 2021-06-17:
Hmm. No, you shouldn't need the grism reffiles to run imaging mode. In imaging mode, Mirage should be making a 2048x2048 dummy transmission image on the fly. Does the AttributeError come after the statements you pasted above? Those statements should be correct when running imaging mode.

micaelabagley, 2021-06-17:
The error comes before the statements I pasted above, and it's actually an OSError (sorry for the AttributeError mention). 
```
OSError: WARNING: unable to read in None
```
So it seems like it's trying to read `None` as a string there?

I've attached the full output:
[log.txt](https://github.com/spacetelescope/mirage/files/6671790/log.txt)

PR #695 (2021-06-17): Allow transmission file to be None or 'None' 
Resolves #693 

This PR allows the transmission filename to be None or 'None' in the input yaml file when working in imaging mode. Previously the code was looking for only None, and was raising a FileNotFound error in cases where the file was listed as 'None'.
closed 2021-06-17 by bhilbert4