spacetelescope/mirage #694: No ra_ref,dec_ref keywords in pointing_dictionary when creating catalogs

micaelabagley, 2021-06-17:
I'm getting a KeyError when using `create_catalog` to query for point sources in my simulated FOV with an input APT pointing and xml file.

I'm using Mirage v2.0.2, and simulating NIRCam Imaging mode.

The issue is that there are no keywords `ra_ref` or `dec_ref` in `pointing_dictionary` (there are `ra` and `dec`).

My call to `create_catalog`:
```
catalog_out = create_catalog.for_proposal(xml_file, pointing_file,
                                          point_source=True, extragalactic=False,
                                          catalog_splitting_threshold=0.12,
                                          out_dir=out_dir, save_catalogs=True)
```
The calls for `ra_ref` and `dec_ref` are in `mirage.catalogs.create_catalogs.for_proposal()`, lines 153-154, 163, 194-195, 216-217.

From what I can tell, the `*_ref` keywords are relevant for the WFSS mode. Is it appropriate for me to replace these `ra_ref` and `dec_ref` with `ra` and `dec`, respectively, in `for_proposal()`?

bhilbert4, 2021-06-17:
Someone else just mentioned this bug to me the other day. Looking at the code it's not clear to me how the ra_ref and dec_ref entries are not being populated. What program is this for? I want to download the APT file and try to recreate the error.

micaelabagley, 2021-06-17:
This is for CEERS (1345), but most of the NIRCam imaging is done in parallel. I've created a separate apt file and copied the pointing info from the CEERS aptx. I'm currently working on a stripped down version with just a single NIRCam pointing (called CEERS5). 

I'm attaching the NIRCam-only apt file (CEERS5_NIRCam.aptx), the xml file (CEERS5_NIRCam.xml), and **edited** pointing file (CEERS5_NIRCam.pointing) and the CEERS pointing file for this pointing (CEERS5.pointing)
[CEERS5.zip](https://github.com/spacetelescope/mirage/files/6673205/CEERS5.zip)

closed 2021-06-18 by bhilbert4

bhilbert4, 2021-06-18:
This should now be fixed in the master branch of the repository. I'm single-parenting tomorrow, but I'll try to find some time to create a new release and put it up on pypi.

PR #696 (2021-06-18): Update for_proposal for recent updates
Resolves #694 

This PR makes a few small changes to `for_proposal()` in create_catalogs.py to account for other recent changes in Mirage. Prior to these changes, `for_proposal()` would crash with missing `ra_ref, dec_ref` entries in the pointing catalog. With this change, `for_proposal()` now calls `ra_dec_update()` in `apt_inputs.py`, which populates ra_ref and dec_ref. With these updates, I was able to successfully create stellar and galactic catalogs using the stripped-down CEERS APT file from #694.