spacetelescope/mirage #699: Can't send images by pasting from clipboard in Arch Linux on Sway

kibybyte, 2021-06-22:
I use Arch Linux and Sway on hardware that fully supports it. I use grim, slurp and wl-copy to take screenshot, save it in my device as well as copy to my clipboard with the following command bound in sway configureation file:
```
#screenshots
bindsym Shift+Print exec grim -c -g "$(slurp)" -t png - | tee /home/$USER/pictures/ss/$(date +"%d-%b-%Y--%H:%M:%S").png | wl-copy
bindsym Print exec grim -c -t png - | tee /home/$USER/pictures/ss/$(date +"%d-%b-%Y--%H:%M:%S").png | wl-copy
```
In every other program I can just paste clipboard contents and send stuff except for mirage. 

Here is a sample of what happens. This just keeps rotating on and on. The program completely freezes if I forcibly click on send-

[image]

I have run from terminal also and gotten these outputs

[image]
[image]

However, If I try to send image by clicking from the UI, it works fine. Even PNG. From this button I mean -

[image]

I love this project and how fast *and* beautiful it is. Kudos to the creators. 

TL;DR: Can't paste clipboard images into mirage running on arch linux on sway

kibybyte, 2021-06-22:
BTW, I get same results regardless of whether I've installed `matrix-mirage` from  the [AUR](https://aur.archlinux.org) or from the [Chaotic AUR](https://aur.chaotic.cx/)

bhilbert4, 2021-06-22:
Unless I'm misunderstanding your question, I think you have created an issue in the wrong repo. Mirage is an astronomical data simulation package.

kibybyte, 2021-06-22:
🤦
 
I am so extremely sorry. This was extremely dumb of me.

kibybyte, 2021-06-22:
is there any way to delete this post?

closed 2021-06-22 by kibybyte

bhilbert4, 2021-06-22:
It's not a problem. I just wanted to make sure that your issue got to the right place.

kibybyte, 2021-06-22:
Thank you and I am once again extremely sorry about this dumb mistake I've made.