spacetelescope/mirage #702: APT with mixture of NIRSpec and NIRCam prime observations

cnaw, 2021-06-29:
Is there an easy way to get mirage to generate the yaml files (and simulated scenes)  when NIRSpec observations are interspersed with others where NIRCam  is the prime instrument  within the same APT proposal ? I have tried feeding the output of proposal 1180 but the code crashes when it arrives at observations where NIRSpec is prime. Thanks!

bhilbert4, 2021-06-30:
This should work. I'll look into it and see what changes need to be made.

cnaw, 2021-06-30:
Thanks @bhilbert4 ! This happens on APT 1180 for instance. Something similar happens on APT 1063 (one of the commissioning tasks) where NIRCam imaging is prime and there is parallel NIRISS imaging.  A quick solution for these cases would be to skip visits where NIRSpec is prime and when NIRCam, NIRISS are prime, skip/ignore the parallels. If that is possible it will be awesome.

bhilbert4, 2021-07-21:
Just a quick note to say that I'm working on this at the moment. Program 1063 completes successfully for me using the current version of Mirage. For 1180, I'm in the process of making some updates. The current function for reading the MOS section of the xml file is missing some options that are present in the program's xml.

cnaw, 2021-07-21:
Cool @bhilbert4 !  git pull should update the code, is that correct ?

bhilbert4, 2021-07-21:
Yep. I just merged the changes, so git pull should get you the updated code.

closed 2021-07-21 by bhilbert4

cnaw, 2021-07-21:
Cool! I will try it out. Thanks a lot!

PR #707 (2021-07-21): Update NRS MOS APT xml reader
This PR updates the NIRSpec MOS mode xml reader to pay attention to the ConfigurationPointing fields, in order to determine the correct order and details of MOS exposures. Testing was done using PID 1180, and everything seems to be working well.

Resolves #702