spacetelescope/mirage #703: NIRCam coronagraphic imaging: unknown pupil

xoraph, 2021-06-30:
I'm trying to synthesize a NIRCam observation with a coronagraph, F210W with MASK210R, from a `fits` file (so far using the `hlsp_candels...` from the example dataset). This is my first time using Mirage, so this may be a user error.
Furthermore, this possibly relates to #612 and #615.

The supposedly relevant part of my input `xml` file (created via `APT`, same pointing as the example data set) reads
```
  <ncc:NircamCoron>
      <ncc:CoronMask>MASK210R</ncc:CoronMask>
      <ncc:AcqTargetID>Same Target as Observation</ncc:AcqTargetID>
      <ncc:AcqTargetBrightness>BRIGHT</ncc:AcqTargetBrightness>
      <ncc:AcqFilter>F210M</ncc:AcqFilter>
      <ncc:AcqReadoutPattern>RAPID</ncc:AcqReadoutPattern>
      <ncc:AcqGroups>3</ncc:AcqGroups>
      <ncc:AcqEtcId>9999.99</ncc:AcqEtcId>
      <ncc:OptionalConfirmationImage>false</ncc:OptionalConfirmationImage>
      <ncc:Subarray>FULL</ncc:Subarray>
      <ncc:DitherPattern>NONE</ncc:DitherPattern>
      <ncc:Filters>
          <ncc:FilterConfig>
              <ncc:Filter>F210M</ncc:Filter>
              <ncc:ReadoutPattern>RAPID</ncc:ReadoutPattern>
              <ncc:Groups>2</ncc:Groups>
              <ncc:Integrations>1</ncc:Integrations>
              <ncc:EtcId></ncc:EtcId>
          </ncc:FilterConfig>
      </ncc:Filters>
      <ncc:PsfReference>true</ncc:PsfReference>
      <ncc:PsfJustification>false</ncc:PsfJustification>
  </ncc:NircamCoron>
```

Using the `Simulated_data_from_mosaic_image.ipynb` example notebook, `mirage.yaml.yaml_generator.SimInput` translates that to a `yaml` file containing
```
  array_name: NRCA2_FULL    # Name of array (FULL, SUB160, SUB64P, etc) overrides subarray_bounds below
  filter: F210M       # Filter of simulated data (F090W, F322W2, etc)
  pupil: MASKRND        # Pupil element for simulated data (CLEAR, GRISMC, etc)
```
(So the apt parsing does not seem to be the issue, hence a new issue.)

When using this for `mirage.seed_image.fits_seed_image.ImgSeed`, I get the following error:
```
[...]/python3.7/site-packages/mirage/utils/flux_cal.py in fluxcal_info(fluxcal_file, instrument, filter_value, pupil_value, detector, module)
    117     if not any(mtch):
    118         raise ValueError(("WARNING: requested filter and pupil values of {} and {} are not in the list of "
--> 119                           "possible options.".format(filter_value, pupil_value)))
    120 
    121     vegazeropoint = zps['VEGAMAG'][mtch][0]

ValueError: WARNING: requested filter and pupil values of F210M and MASKRND are not in the list of possible options.
```
As evident from the error message, this occurs because `MASKRND` is not listed as a pupil (or at all) in `mirage/config/NIRCam_zeropoints.list` and `mirage/config/nircam_filter_pupil_pairings.list`. (`MASK*` entries exist in `mirage/config/NIRCam_subarray_definitions.list` and `mirage/config/nircam_filter_and_pupil_wheel_positions.txt`.)

I could try and circumvent that by manually setting `pupil: "CLEAR"` in the `yaml` file, but that would go against the allowed combinations listed [here](https://jwst-docs.stsci.edu/near-infrared-camera/nircam-instrumentation/nircam-pupil-and-filter-wheels#NIRCamPupilandFilterWheels-Allowedcombinationsbetweenpupilandfilterwheelelements). (It also leads to a shape mismatch between `dark` and `ImgSeed` in `mirage.ramp_generator.obs_generator.Observation`, because `siaf_interface` determines the `dark` dimensions to be 128x128, not 2048x2048.)

If this is a user error, maybe adding a short explanation of the correct procedure in one of the imaging notebooks could help prevent this from occurring to other users.
If I have overlooked the corresponding issue including a manual fix, please point me to it and close/delete this one.
Else, a fix (or a quick instruction how to manually deal with this) would be highly appreciated.

mperrin, 2021-06-30:
Hi @xoraph  - FYI, MIRAGE does not really support coronagraphy correctly, not out of the box. However @kammerje has been doing some work to develop an implementation for that, combining MIRAGE with the pyNRC simulator. You may find it useful to coordinate with him.

xoraph, 2021-06-30:
Hi @mperrin, thanks a lot for this info!
I will go back to synthetic imaging without a coronagraph for now (adding a coronagraph would have been icing on the cake), and will get in touch with @kammerje as soon as my problem set mandates it.

Closing this issue.

closed 2021-06-30 by xoraph