spacetelescope/mirage #708: Inconsistent source catalog entries

bhilbert4, 2021-07-29:
From @obi-wan76 :

I’ve been looking at some discrepancies in the MIRAGE catalog that you may be able to provide some clarification. I’ll try to summarized the problem with an example:
 
For our program Fine Phasing 3 (PID 1160) we are using a target star with the following parameters:
TYC-8902-1241-1
k = 8.918
ra = '06 25 25.0836'
dec = '-63 56 41.59'
 
When running create_catalog.get_all_catalogs with a large radius (7’ à 420”) I get the target from the catalog to be:
2
96.354515
-63.944885
9.66733
With nircam_f212n_clear_magnitude = 9.66733. The coordinates are consistent with the target star but, to me,  a SW F212N magnitude of 9.67 seem too faint for a K = 8.918 star.  So, I ran create_catalog.get_all_catalogs with a different search radius of 10”, so to isolate the target star,  from which I got a single source:
1
96.354515
-63.944885
8.9304695
 
The same coordinates as before but with a different SW F212N magnitude of 8.93, which seems more consistent for a K=8.918 star. However, this 8.93 mag source doesn’t show up in the larger catalog (7’, above).
 
If I increase the search radius to 1’ then I see both matching sources (9.66 & 8.93 mag):
96.354515
-63.944885
9.66733
96.361626
-63.952843
8.9304695
 
However, this time they swapped the coordinates. At some point, if I keep increasing the search radius the 8.93 mag source dropped out from the catalog all together.
 
So, I guess the question is, what happened to the 8.93 mag source? That source seems to be a more reasonable match to the target star with a K = 8.918.
 
Something similar happened with another of our programs where the “best” match source disappeared in the larger catalog (or as we increased the search radius). My concern is that during our WF team practices we have our K mag requirements for target selection and SBR/SN values but it seems that for some of these sources we may be simulating a target fainter than we should expect.

obi-wan76, 2021-07-29:
If I do an even smaller search radius, 0.5”, then I get both sources with a very small positional difference.
 
1	96.354515	-63.944885	9.66733
2	96.35459	-63.944935	8.9304695							
 
A search radius of 0.4” also gives two sources:
 
1	96.354515	-63.944885	9.66733
2	96.35459	-63.944935	8.9304695							
 
However, a search radius of 0.2 and 0.3” gives one source, the fainter one:
1	96.354515	-63.944885	9.66733

bhilbert4, 2021-08-06:
I'm starting to think this is a problem with the source matching and/or cross-referencing. https://github.com/spacetelescope/mirage/blob/master/mirage/catalogs/create_catalog.py#L1223

I've been playing with running get_all_catalogs with a range of box widths, centered on the target coordinates above. I'm seeing that LW magnitude values for the target do not seem to change with box width. But shorter filters (F090W, F200W) do have changing magnitudes. From the description of the function above, the magnitude calculation differs depending on whether or not a matching source from 2MASS is present. Also, there is a 0.3 arcsec maximum threshold for source matching. I have a feeling that these two details are causing different answers depending on exactly what sources are returned in the queries.

We could tighten up the maximum offset (currently 0.3"), although at some point this might end up causing good matches to be rejected, depending on the precision of the various input catalogs. It'll take me some time to fully digest what the cross matching function is doing. Maybe there are some other tweaks we can make as well.

bhilbert4, 2021-10-05:
Further investigation with the bugfix in #726 in place:
I still see a small number of sources with inconsistent brightnesses/positions depending on the area queried.

Looking closely at one of these sources:

Source is located ~57" from the center of the search box:
For a 100"x100" search box, the source is not present in the 2MASS or WISE query results, but is present in GAIA results. This is because of the difference in the source's location between the catalogs. In GAIA, the source's location places it just inside the search area, by a fraction of an arcsecond. For 2MASS and WISE, the source's location places it just outside the query area. So in this case, the source does appear in the final source catalog, but the position is that from GAIA, and the brightnesses are interpolated from the magnitudes returned by GAIA only.

Looking at the same source, but with a 400"x400" search box:
In this case the source is present in all 3 catalog query results. The way the code is currently set up, the source location in the final catalog begins as that  from GAIA. If the source is present in 2MASS, then that location is used, and finally, if the source is present in WISE, then that location is used. This means that in this case, the source's final position is slightly shifted from what it is in the 100"x100" search case, since WISE uses a slightly different location than GAIA. Also, in this case, the magnitudes of the source are determined by interpolating from the GAIA, 2MASS, and WISE magnitudes, rather than just the GAIA magnitudes. 

I don't think there's much that can be done to fix cases like this. There will always be small inconsistencies around the edges of the queried area. My recommendation would be to perform catalog queries using an area on they sky that is comfortably larger than the area covered by the detectors being simulated. How much bigger? The catalog with the highest positional uncertainties is 2MASS, with mean uncertainties of ~0.6-0.7". So it seems like as long as your query area is larger than the detector FOV by several arcseconds, you should be safe.

Also, the astrometric uncertainties are a lot smaller for WISE and Gaia. We should re-examine the code to be sure and if necessary, default to using source locations from one of these two catalogs, rather than 2MASS.