spacetelescope/mirage #729: Broadcast ValueError at combine_seeds

eas342, 2021-09-17:
I am running a NIRCam grism TSO simulations with 6 segments and am getting an error with the combine_seeds step for the 5th segment.

I ran the following with https://github.com/eas342/mirage_runs
```python
python NIRCam_grism_tso_wrapper.py mirage_input_004_manatee/source_params_wasp80.yaml
```

The error traceback is a s follows
```
Traceback (most recent call last):
  File "NIRCam_grism_tso_wrapper.py", line 936, in <module>
    gW.do_all()
  File "NIRCam_grism_tso_wrapper.py", line 926, in do_all
    self.create_sim()
  File "NIRCam_grism_tso_wrapper.py", line 879, in create_sim
    gr_tso.create()
  File "/home/u17/eas342/miniconda2/envs/mirage/lib/python3.7/site-packages/mirage/grism_tso_simulator.py", line 591, in create
    obs.create()
  File "/home/u17/eas342/miniconda2/envs/mirage/lib/python3.7/site-packages/mirage/logging/logging_functions.py", line 111, in wrapped
    func(*args, **kwargs)
  File "/home/u17/eas342/miniconda2/envs/mirage/lib/python3.7/site-packages/mirage/ramp_generator/obs_generator.py", line 1226, in create
    self.seed_image, self.segmap, self.seedheader = self.combine_seeds(seed_files)
  File "/home/u17/eas342/miniconda2/envs/mirage/lib/python3.7/site-packages/mirage/ramp_generator/obs_generator.py", line 966, in combine_seeds
    seed[int_start: int_start+ints, grp_start: grp_start+groups, :, :] = seed_data
ValueError: could not broadcast input array from shape (61,12,256,2048) into shape (56,12,256,2048)

```
The full super long traceback is at:
https://arizona.box.com/s/5c2pdu67y7fh2egdimeuknnx3cyg62om

Could this be related to https://github.com/spacetelescope/mirage/issues/677 ?

eas342, 2021-09-19:
I have a suspicion that this could be related to the co-adding step. This is because I repeated the same simulation with RAPID read mode and twice the number of groups and it ran through without problems.

bhilbert4, 2021-09-20:
I think you may be right. What are the values of the PTINTSRT and PTFRMSRT header keywords in the seed image files?

eas342, 2021-09-20:
Here's what they are:

```
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg001_part001_seed_image.fits
PTINTSRT=0, PTFRMSRT=0
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg001_part002_seed_image.fits
PTINTSRT=106, PTFRMSRT=0
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg002_part001_seed_image.fits
PTINTSRT=0, PTFRMSRT=0
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg002_part002_seed_image.fits
PTINTSRT=106, PTFRMSRT=0
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg003_part001_seed_image.fits
PTINTSRT=0, PTFRMSRT=0
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg003_part002_seed_image.fits
PTINTSRT=106, PTFRMSRT=0
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg004_part001_seed_image.fits
PTINTSRT=0, PTFRMSRT=0
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg004_part002_seed_image.fits
PTINTSRT=106, PTFRMSRT=0
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg005_part001_seed_image.fits
PTINTSRT=0, PTFRMSRT=0
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg005_part002_seed_image.fits
PTINTSRT=106, PTFRMSRT=0
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg006_part001_seed_image.fits
PTINTSRT=0, PTFRMSRT=0
Filename = jw01185002001_01101_00002_nrca5_uncal_F322W2_seg006_part002_seed_image.fits
PTINTSRT=106, PTFRMSRT=0
```

eas342, 2021-09-28:
It looks like you may already have a solution with pull request https://github.com/spacetelescope/mirage/pull/731
However, if it's of any use, I noticed today that the Part 1 and Part 2 of the last segment disagreed about the segment end.
```bash
# HDU 0 in jw01185002001_01101_00002_nrca5_uncal_F322W2_seg004_part001_seed_image.fits:
SEGINTED=                  741                                                  
# HDU 0 in jw01185002001_01101_00002_nrca5_uncal_F322W2_seg004_part002_seed_image.fits:
SEGINTED=                  741                                                  
# HDU 0 in jw01185002001_01101_00002_nrca5_uncal_F322W2_seg005_part001_seed_image.fits:
SEGINTED=                  953                                                  
# HDU 0 in jw01185002001_01101_00002_nrca5_uncal_F322W2_seg005_part002_seed_image.fits:
SEGINTED=                  953                                                  
# HDU 0 in jw01185002001_01101_00002_nrca5_uncal_F322W2_seg006_part001_seed_image.fits:
SEGINTED=                 1165                                                  
# HDU 0 in jw01185002001_01101_00002_nrca5_uncal_F322W2_seg006_part002_seed_image.fits:
SEGINTED=                 1120         
```

closed 2021-10-12 by bhilbert4