spacetelescope/mirage #764: no dark current subtraction

x12red, 2022-02-06:
Hi all, 

I've created some simulated exposures with MIRAGE.
When I try to run the first step of the JWST pipeline, the header file of the output fits file (rate.fits) show me that the dark subtraction step has not been performed.  Does it happen because of an error in simulating exposures with MIRAGE?

Thanks in advance

bhilbert4, 2022-02-07:
The default is to subtract the dark current though, so you should have to explicitly turn off dark current subtraction in order for it to be skipped. This should complete all steps:

```
from jwst.pipeline.calwebb_detector1 import Detector1Pipeline
p = Detector1Pipeline()
p.save_results = True
p.run('my_file_uncal.fits')
```

x12red, 2022-02-09:
Thanks for your reply!

Yes, it is a pipeline issue. It seems that the pipeline does not contain reference files optimised for what I simulated with MIRAGE.  The JWST helpdesk told me that we could create simulated dark exposures to work around this problem and use them as reference files. 

So, maybe, we could use the dark linearized objects that MIRAGE creates after we simulate an exposure. Does it make sense?

bhilbert4, 2022-02-09:
The dark current reference files currently in use for the NIRCam shortwave detectors are all zeros. That's because the NIRCam team did not collect enough darks during ground testing to produce a high signal to noise dark current reference file (because the dark rate is very low). In theory you could average together the dark current files that Mirage uses to create simulations. However there are so few of these that the resulting mean dark current ramp would still end up not doing much besides adding noise to your data after you subtract it. I recommend skipping dark subtraction for the moment, or just let the pipeline subtract the current file full of zeros. Good dark current reference files will be constructed and delivered by the NIRCam team later in commissioning.

closed 2022-02-17 by bhilbert4