spacetelescope/mirage #769: Update MIRAGE and the JWST pipeline

x12red, 2022-02-24:
Hi all, 

Currently, I am working with MIRAGE 2.1.0 and the JWST pipeline 1.1.0. 
I would like to work with the latest version of the JWST pipeline, but it seems that it does not accept the uncalibrated files generated by MIRAGE 2.1.0. 

Is there a way I can update the above package to work with them without any software incompatibility?
Should I simply create another conda env?

bhilbert4, 2022-02-24:
Were you able to resolve this? I see the response to your question in https://github.com/spacetelescope/jwst/issues/6742#issuecomment-1049996178

x12red, 2022-02-27:
Yes, I solved it following what has been suggested in the question you mentioned above

closed 2022-02-28 by bhilbert4