spacetelescope/mirage #788: CRDS Error - NIRISS SOSS Exoplanet Notebook

tgrindlay, 2022-04-25:
Hello there! 

I am a student new to exploring MIRAGE, I attempted to try out the NIRISS SOSS exoplanet example notebook.  When I performed the sim.create() command I received the following error: 

CRDS - ERROR -  (FATAL) CRDS server connection and cache load FAILED.  Cannot continue.
See https://hst-crds.stsci.edu/docs/cmdline_bestrefs/ or https://jwst-crds.stsci.edu/docs/cmdline_bestrefs/
for more information on configuring CRDS,  particularly CRDS_PATH and CRDS_SERVER_URL. : [Errno 2] No such file or directory: '/grp/crds/cache/config/jwst/server_config'

I followed the second link to the JWST CRDS install/config page and tried to follow the steps. However, I am currently stuck in the “Assigning, obtaining, and caching reference files” section, as the Performing JWST Calibrations link leads to a 404 Not Found error. 

I would appreciate any advice or input you have on how to remedy this.
Let me know if this is the right place to ask these questions, if not, I would appreciate directions to the correct procedure. 
Thanks!

bhilbert4, 2022-04-25:
Since CRDS is providing an HST link rather than JWST, my guess is that your environment variables are not set to the right values.
Do you have them pointing to the locations in the [documentation](https://mirage-data-simulator.readthedocs.io/en/latest/reference_files.html#crds-environment-variables)?

```
    export CRDS_PATH=$HOME/crds_cache
    export CRDS_SERVER_URL=https://jwst-crds.stsci.edu
```

Bryan

tgrindlay, 2022-04-25:
Hi Bryan!

I believe I have done this? I have a bash cell (%%bash) with these commands within the notebook, is this a valid way to establish those environment variables?

bhilbert4, 2022-04-25:
It may be. I don't think I've ever tried that method. I usually set environment variables using the `os` package.
```
os.environ["CRDS_PATH"] = "$HOME/crds_cache"
os.environ["CRDS_SERVER_URL] = "https://jwst-crds.stsci.edu"
```

Once those are set up, you shouldn't have to think much about reference files. They'll be downloaded to your local cache at the time that you (or Mirage) calls the calibration pipeline. I'd suggest trying this method and see if you get the same error.

tgrindlay, 2022-04-25:
That has removed my error, thanks so much! It begins fetching and downloading files but my kernel dies each time at this point: 

```
Simulating WASP43 in WASP43 transit observation
Configuration: SUBSTRIP256 + CLEAR
Groups: 6, Integrations: 320
Constructing frames for chunk 1/40...
```
Then I receive the kernel restarting message pop up.
Is there any way to prevent this?

bhilbert4, 2022-04-26:
I don't think I've ever seen that before. @hover2pi have you? I wonder if you were to copy/paste the cells from the notebook into a script and run it, if it would work? Or perhaps give a more helpful message when it fails?

tgrindlay, 2022-04-26:
Since I'm using a virtual environment for mirage, and this is through an online Jupyter server, I made the py file script on my computer and then ran it through terminal in Jupyter with the command `python3 FILENAME.py` . This appeared to be working until it killed the command at the same point. I'll put all the messages given while this is running in case that clarifies anything: 

```
**WARNING**: LOCAL JWST PRD VERSION PRDOPSSOC-044 DOESN'T MATCH THE CURRENT ONLINE VERSION PRDOPSSOC-045
Please consider updating pysiaf, e.g. pip install --upgrade pysiaf or conda update pysiaf
No module named 'exoctk'
There was a problem computing those limb darkening coefficients. Using all zeros.
No module named 'exoctk'
There was a problem computing those limb darkening coefficients. Using all zeros.
No PSF files detected. Using all ones.
No PSF files detected. Using all ones.
2022-04-26 18:04:40,199 - mirage.soss_simulator - INFO - 

Running soss_simulator....

2022-04-26 18:04:40,199 - stpipe - INFO - 

Running soss_simulator....

2022-04-26 18:04:40,199 - stpipe - INFO - 

Running soss_simulator....

2022-04-26 18:04:40,200 - mirage.soss_simulator - INFO - Using parameter file: 
2022-04-26 18:04:40,200 - stpipe - INFO - Using parameter file: 
2022-04-26 18:04:40,200 - stpipe - INFO - Using parameter file: 
2022-04-26 18:04:40,200 - mirage.soss_simulator - INFO - niriss_soss_substrip256_clear.yaml
2022-04-26 18:04:40,200 - stpipe - INFO - niriss_soss_substrip256_clear.yaml
2022-04-26 18:04:40,200 - stpipe - INFO - niriss_soss_substrip256_clear.yaml
Simulating WASP43 in WASP43 transit observation
Configuration: SUBSTRIP256 + CLEAR
Groups: 6, Integrations: 320

Constructing frames for chunk 1/40...
Killed
```