spacetelescope/mirage #794: key error "designation" from create_catalog.for_proposal

christinawilliams, 2022-05-10:
Hi! I'm having an issue with mirage's create_catalogs routines returning an error "KeyError 'designation'" when I call create_catalog.for_proposal. It is a command from the online examples that previously ran successfully in October, and I have not updated any of the packages since then (I also have a fresh install / update on a different machine and get the same error).

bhilbert4, 2022-05-10:
You're getting the same error for a new Mirage installation? I just merged a PR at the end of March because astroquery changed the names of some of their catalog columns from 'designation' to 'DESIGNATION'. Your error makes it seem like you're still using a version of the code from before the change?

christinawilliams, 2022-05-27:
> Yes, its "new" meaning I installed it in January but I just updated it
> today using pip install --update mirage (it is version 2.2.1). But I
> haven't updated astroquery in a long time, could that be the issue?

I'm running the latest 0.4.6 version of astroquery after updating (MIRAGE
is latest version 2.2.1) and still having this exact same error. Any other
ideas?

bhilbert4, 2022-05-27:
This is what I'm getting now, using astroquery 0.4.6:

```
2022-05-27 16:07:31,907 - mirage.apt.apt_inputs - INFO - csv exposure list written to /Users/hilbert/python_repos/test_mirage_christina/Observation_table_for_1963.csv
2022-05-27 16:07:40,490 - mirage.catalogs.create_catalog.query_GAIA_ptsrc_catalog - INFO - Searching the GAIA DR2 catalog
2022-05-27 16:07:43,359 - mirage.catalogs.create_catalog.query_GAIA_ptsrc_catalog - INFO - Retrieved 20 sources for catalog gaia
2022-05-27 16:07:45,164 - mirage.catalogs.create_catalog.query_GAIA_ptsrc_catalog - INFO - Retrieved 6 sources for catalog tmass
2022-05-27 16:07:46,821 - mirage.catalogs.create_catalog.query_GAIA_ptsrc_catalog - INFO - Retrieved 153 sources for catalog wise
2022-05-27 16:07:49,612 - mirage.catalogs.create_catalog.query_GAIA_ptsrc_catalog - INFO - Retrieved 6 sources for catalog tmass_crossmatch
2022-05-27 16:07:52,478 - mirage.catalogs.create_catalog.query_GAIA_ptsrc_catalog - INFO - Retrieved 16 sources for catalog wise_crossmatch
Traceback (most recent call last):
  File "make_cat.py", line 16, in <module>
    ptsrcs, galaxies, ptsrc_files, galaxy_files, ptsrc_mapping, gal_mapping = create_catalog.for_proposal(xml, pointing, out_dir='./',
  File "/Users/hilbert/python_repos/mirage/mirage/catalogs/create_catalog.py", line 282, in for_proposal
    tmp_cat, tmp_filters = get_all_catalogs(mean_ra, mean_dec, full_width,
  File "/Users/hilbert/python_repos/mirage/mirage/catalogs/create_catalog.py", line 654, in get_all_catalogs
    twomass_cat, twomass_cols = query_2MASS_ptsrc_catalog(outra, outdec, box_width)
  File "/Users/hilbert/python_repos/mirage/mirage/catalogs/create_catalog.py", line 382, in query_2MASS_ptsrc_catalog
    query_table = Irsa.query_region(ra_dec_string, catalog='fp_psc', spatial='Box',
  File "/Users/hilbert/miniconda3/envs/mirage-py3.8/lib/python3.8/site-packages/astroquery/utils/class_or_instance.py", line 25, in f
    return self.fn(obj, *args, **kwds)
  File "/Users/hilbert/miniconda3/envs/mirage-py3.8/lib/python3.8/site-packages/astroquery/utils/process_asyncs.py", line 26, in newmethod
    response = getattr(self, async_method_name)(*args, **kwargs)
  File "/Users/hilbert/miniconda3/envs/mirage-py3.8/lib/python3.8/site-packages/astroquery/ipac/irsa/core.py", line 191, in query_region_async
    response = self._request("GET", url=Irsa.IRSA_URL,
  File "/Users/hilbert/miniconda3/envs/mirage-py3.8/lib/python3.8/site-packages/astroquery/query.py", line 315, in _request
    response = query.from_cache(self.cache_location)
  File "/Users/hilbert/miniconda3/envs/mirage-py3.8/lib/python3.8/site-packages/astroquery/query.py", line 114, in from_cache
    response = pickle.load(f)
ModuleNotFoundError: No module named 'astroquery.irsa.core'
```

christinawilliams, 2022-10-11:
Yes, its "new" meaning I installed it in January but I just updated it
today using pip install --update mirage (it is version 2.2.1). But I
haven't updated astroquery in a long time, could that be the issue?

bhilbert4, 2022-10-11:
That could be the issue. astroquery just updated the catalog column name fairly recently. So if you have an old version of astroquery but a new version of Mirage, they may be looking for different column names. Try upgrading astroquery to version 0.4.6.