spacetelescope/mirage #795: Error in catalog_seed_image.py that crashes mirage

christinawilliams, 2022-05-12:
I'm trying to use a self-made input catalog of simulated galaxies for NIRCam using MIRAGE, and am getting an error that I think crashes inside catalog_seed_image.py. I used the examples on the github page for making your own 2D sersic profile galaxy input catalogs, and so I think they are in the correct format using catalog_generator.GalaxyCatalog and add_magnitude_column. I attached the last 3 lines of output before I get all the code traceback:

 File "/home/cwilliams/anaconda3/envs/JWST/lib/python3.8/site-packages/mirage/seed_image/catalog_seed_image.py", line 3720, in RADecToXY_astrometric
    pixelx, pixely = self.coord_transform.inverse(loc_v2, loc_v3)
AttributeError: 'TaggedDict' object has no attribute 'inverse'

I am getting this same error when using two different mirage installations, 2.1.1.dev48+g9bdb234 and 2.2.1.

bhilbert4, 2022-05-13:
What do you have listed for the Reffiles:astrometric entry in your yaml file? It sounds like it may not be pointing to an asdf-formatted reference file?

christinawilliams, 2022-10-11:
My yaml file doesn't seem to have that entry at all.