spacetelescope/mirage #797: Error extracting reference files using downloader

christinawilliams, 2022-05-13:
When trying to download the reference files using downloader, I get an error:
EOFError: Compressed file ended before the end-of-stream marker was reached
from the command: downloader.download_reffiles(download_path, instrument='NIRCam', dark_type='linearized', skip_darks=False, single_dark=False, skip_cosmic_rays=False, skip_psfs=False)
I think it is after the download is complete, the error backtracks to --> 323     file_object.extractall(path=directory)

bhilbert4, 2022-05-13:
Was there any indication of what file it was working on when it failed? I can start up the same command, but given the size of the darks, it may take a while to get a result. My initial guess is that maybe the download was interrupted somehow resulting in a partial/corrupted file? But that's just a guess.

christinawilliams, 2022-05-20:
No there wasn't from MIRAGE, but I tried a few times to delete any file
that looked suspiciously small and re-ran the downloader. Eventually this
solved the issue, thank you!

christinawilliams, 2022-05-20:
Actually, it is not solved, I am still having issues with
downloader.download_reffiles. This time for instrument="NIRISS", it crashes
with this error:
RuntimeError: Wrong URL -
https://data.science.stsci.edu/redirect/JWST/jwst-simulations/mirage_reference_files/niriss/darks/linearized/Linearized_Dark_and_SBRefpix_NISNIRISSDARK-153451235_17_496_SE_2015-12-11T17h59m52_dms_uncal_linear_dark_prep_object.fits.gz
I also wasn't able to use the skip_grism keyword which was used in one of
MIRAGE examples I found online. Any ideas?

bhilbert4, 2022-05-23:
I just tried to download that NIRISS dark and didn't have any trouble. Maybe you had some sort of internet connection hiccup at just the wrong time?

As for the skip_grism keyword, I stripped that out of the code a while back when the grism-specific reference files were removed from Mirage (they're now downloaded from Nor's GRISM_NIRCAM and GRISM_NIRISS repositories. I'll remove the keyword from the documentation. Sorry about that.

closed 2022-05-24 by bhilbert4