spacetelescope/mirage #799: MIRAGE sensitivity files: what wavelength binning?

tomgreene, 2022-05-24:
I am interested in using MIRAGE to simulate slitless spectra for another mission (Pandora smallsat).

I have a question about how to make the needed sensitivity files: over what bandpass is the instrument signal binned when computing the electrons s^-1 / (erg / s/ cm^2 / Ang) ? 
Specifically, is the electrons s^-1 computed per spectral pixel or is it binned spectrally some other way? 
I assume that there is no spatial info in these files (i.e., sensitivity is integrated over all spatial pixels), and please correct me if that is wrong.

Thanks in advance for any info.

tomgreene, 2022-05-26:
Thanks @bhilbert4 .

I have been looking at the MIRAGE NIRCam sensitivity file values, and it looks like they may be in e-/s/A per erg/s/cm^2/A , the same units used in aXeSIM simulations. @NorPirzkal please let me know if this is incorrect.

NorPirzkal, 2022-05-26:
The sensitivities used in Mirage have been in e-/s, that is correct.